ArduPilot/apm_planner #574: apm_planner Permanent flight plan artifact on map

labtoast, 2015-01-06:
On application start, if there is a flight plan displayed on the map when apm_planner before switching to "flight plan" view, when switching to flight plan view an artifact of the plan is generated and displayed permanently.  

Ubuntu 14.04

Steps to reproduce:
1) start SITL vehicle
2) start apm_planner
3) add waypoints onto map view (not necessary if path loaded from vehicle that you can see)
4) click "Flight Plan" tab

log: http://pastebin.com/izKRLTuv

[image]

billbonney, 2015-01-06:
I have the same issue with SITL connections. I haven't seen it with a serial connection using USB or radios

labtoast, 2015-01-06:
I think that any UDP connected vehicle will cause this issue to occur.

alternative steps without SITL:
1) send dummy heartbeats over UDP connection
2) start apm_planner
3) add waypoints onto map view
4) click "Flight Plan" tab

code: http://pastebin.com/UeBXjYRR

Arne-W, 2017-04-05:
This issue will be fixed with #1025. As there is an duplicate of this bug #1024 which is referenced by the PR I will close this issue.

closed 2017-04-05 by Arne-W