ArduPilot/apm_planner #579: Radio config: some parameters not dislayed

AndKe, 2015-01-10:
[image]

billbonney, 2015-01-10:
Is this a 433MHz radio?

AndKe, 2015-01-10:
Yes, the standard, stock HM-TRP based 3DR radio.  I am just running custom firmware for additional security, so that's why you see an "s" after SiK 1.9

billbonney, 2015-01-10:
I only have a 915MHz to test with. I'll see how I can solve that

AndKe, 2015-01-10:
Feel free to tell me how to test, or maybe add some logging/debugging lines.

billbonney, 2015-01-11:
If you can connect to the radio in a terminal windows and connect to it using screen command and type ATI5 to get a param dump that would help. you type +++ to enter command mode, then type ATI5

```
$ screen /dev/cu.usbserial-AH016D6C 57600

OK
ATI5
S0:FORMAT=25
S1:SERIAL_SPEED=57
S2:AIR_SPEED=64
S3:NETID=42
S4:TXPOWER=20
S5:ECC=1
S6:MAVLINK=1
S7:OPPRESEND=1
S8:MIN_FREQ=915000
S9:MAX_FREQ=928000
S10:NUM_CHANNELS=50
S11:DUTY_CYCLE=100
S12:LBT_RSSI=0
S13:MANCHESTER=0
S14:RTSCTS=0
S15:MAX_WINDOW=131
```

AndKe, 2015-01-11:
OK                                                                                                 
ATI5                                                                                               
S0:FORMAT=25                                                                                       
S1:SERIAL_SPEED=57                                                                                 
S2:AIR_SPEED=64                                                                                    
S3:NETID=0                                                                                         
S4:TXPOWER=20                                                                                      
S5:ECC=1                                                                                           
S6:MAVLINK=1                                                                                       
S7:OPPRESEND=1                                                                                     
S8:MIN_FREQ=433050                                                                                 
S9:MAX_FREQ=434790
S10:NUM_CHANNELS=10
S11:DUTY_CYCLE=100
S12:LBT_RSSI=0
S13:MANCHESTER=0
S14:RTSCTS=1
S15:MAX_WINDOW=131

billbonney, 2015-01-11:
can you send me the log line that corresponds to one below. It should read 67 (ie. 0x43) for 433Mhz
DEBUG 2015-01-11T09:38:02.250 Read Local Freq: 145 

Thanks

AndKe, 2015-01-11:
I am sending more of it, because I think you'll find it useful:
```
DEBUG 2015-01-11T20:18:40.675 Verison Compare: " New Version 2.0.14 compared to Old Version 2.0.16" 
 INFO 2015-01-11T20:18:40.676 no new update available 
DEBUG 2015-01-11T20:18:55.821 3DR Radio Start Serial Port Scanning 
 INFO 2015-01-11T20:18:58.399 read 3DR Radio Settings 
DEBUG 2015-01-11T20:18:58.407 Radio3DRSettings Serial Port Open SUCCESS "ttyUSB0" 
DEBUG 2015-01-11T20:18:58.408 Port "ttyUSB0" 57600 8 FC 0 P 0 SB 1 
 INFO 2015-01-11T20:18:58.410 Port Configured with success 
DEBUG 2015-01-11T20:18:58.411 Radio3DRSettings::writeEscapeSeqeunce() 
DEBUG 2015-01-11T20:18:58.411 local status: "entering command mode" 
 INFO 2015-01-11T20:18:58.412 Retry  1 
DEBUG 2015-01-11T20:18:59.535 Radio3DRSettings::writeEscapeSeqeunceNow() 
DEBUG 2015-01-11T20:19:00.525 Radio readData currentline: "OK
" 
 INFO 2015-01-11T20:19:00.526 3DR Radio: Entered Command Mode 
DEBUG 2015-01-11T20:19:00.527 local status: "Entered Command mode" 
DEBUG 2015-01-11T20:19:00.528 Radio3DRSettings::readLocalVersionString() 
DEBUG 2015-01-11T20:19:00.529 local status: "Read local radio version" 
DEBUG 2015-01-11T20:19:00.538 Radio readData currentline: "ATI0
" 
DEBUG 2015-01-11T20:19:00.539 Radio readData nextline: "SiK 1.9FFN on HM-TRP
" 
 INFO 2015-01-11T20:19:00.540 3DR Local Radio: Version "SiK 1.9FFN on HM-TRP
" 
DEBUG 2015-01-11T20:19:00.541 Radio3DREeprom Ver: "SiK 1.9FFN on HM-TRP
" 
DEBUG 2015-01-11T20:19:00.542 local status: "Read local radio version" 
DEBUG 2015-01-11T20:19:00.543 Radio3DRSettings::readLocalRadioFrequency() 
DEBUG 2015-01-11T20:19:00.544 local status: "Read local radio frequency" 
DEBUG 2015-01-11T20:19:00.550 Radio readData currentline: "ATI3
" 
DEBUG 2015-01-11T20:19:00.551 Radio readData nextline: "67
" 
DEBUG 2015-01-11T20:19:00.552 Read Local Freq: 67 
DEBUG 2015-01-11T20:19:00.553 local status: "read local radio frequency" 
DEBUG 2015-01-11T20:19:00.553 Radio3DRSettings::readLocalSettingsStrings() 
DEBUG 2015-01-11T20:19:00.554 local status: "Read local radio eeprom values" 
DEBUG 2015-01-11T20:19:00.562 Radio readData currentline: "ATI5
" 
DEBUG 2015-01-11T20:19:00.563 Radio readData nextline: "S0:FORMAT=25
" 
DEBUG 2015-01-11T20:19:00.564 Radio3DREeprom Param: "S0:FORMAT=25
" 
DEBUG 2015-01-11T20:19:00.565 Radio readData currentline: "S1:SERIAL_SPEED=57
" 
DEBUG 2015-01-11T20:19:00.565 Radio3DREeprom Param: "S1:SERIAL_SPEED=57
" 
DEBUG 2015-01-11T20:19:00.567 Radio readData currentline: "S2:AIR_SPEED=64
" 
DEBUG 2015-01-11T20:19:00.567 Radio3DREeprom Param: "S2:AIR_SPEED=64
" 
DEBUG 2015-01-11T20:19:00.569 Radio readData currentline: "S3:NETID=390
" 
DEBUG 2015-01-11T20:19:00.569 Radio3DREeprom Param: "S3:NETID=390
" 
DEBUG 2015-01-11T20:19:00.571 Radio readData currentline: "S4:TXPOWER=20
" 
DEBUG 2015-01-11T20:19:00.572 Radio3DREeprom Param: "S4:TXPOWER=20
" 
DEBUG 2015-01-11T20:19:00.574 Radio readData currentline: "S5:ECC=1
" 
DEBUG 2015-01-11T20:19:00.575 Radio3DREeprom Param: "S5:ECC=1
" 
DEBUG 2015-01-11T20:19:00.576 Radio readData currentline: "S6:MAVLINK=1
" 
DEBUG 2015-01-11T20:19:00.577 Radio3DREeprom Param: "S6:MAVLINK=1
" 
DEBUG 2015-01-11T20:19:00.578 Radio readData currentline: "S7:OPPRESEND=1
" 
DEBUG 2015-01-11T20:19:00.579 Radio3DREeprom Param: "S7:OPPRESEND=1
" 
DEBUG 2015-01-11T20:19:00.582 Radio readData currentline: "S8:MIN_FREQ=433050
" 
DEBUG 2015-01-11T20:19:00.582 Radio3DREeprom Param: "S8:MIN_FREQ=433050
" 
DEBUG 2015-01-11T20:19:00.585 Radio readData currentline: "S9:MAX_FREQ=434790
" 
DEBUG 2015-01-11T20:19:00.585 Radio3DREeprom Param: "S9:MAX_FREQ=434790
" 
```

also note the additional lineshift or CR before the final " on these lines.

closed 2015-02-06 by billbonney

AndKe, 2015-02-06:
please note that HW Flow control is still not indicated properly (the droplist does not select "on" when so)
HW Flow could very well be a checkbox like ECC and OP_Resend.

billbonney, 2015-02-06:
I fixed the issue of it not updating.

commit 3a8236c4 (2015-02-06): 3DR Radio Config: Fixes #579 433MHz frequencies not showing correctly