ArduPilot/apm_planner #584: _xgeWireToEvent: Unknown extension 148, this should never happen.

AndKe, 2015-01-15:
Whenever I run APM Planner, it fills mu console with 
"_xgeWireToEvent: Unknown extension 148, this should never happen."
even worse, I fill the /.cache/upstart (log cache) with 16GB in minutes. 
Lots of ssd activity, and wasted power.
Please try to find out what causes it.

billbonney, 2015-01-15:
A quick search does not turn up a solution to this. My advice would be to do a complete clean and rebuild and make sure nothing in the build has gone wonky.

AndKe, 2015-01-16:
This time, I compiled thru Eclipse just to organize output.
I posted errors,warnings and info section here: 
https://gist.github.com/AndKe/7e99746d4f79907010bc

Would you please take a look to see if you can spot anything that could be related to this problem ?

AndKe, 2015-01-16:
I did some research.
The pesky log event is caused by some applications.  Like Seam client, there were speculations about a related application crash too.
 
The ablility to create these lines got removed from latest libXext.so.6.4.0  http://cgit.freedesktop.org/xorg/lib/libXext/commit/?id=11aad96bd689d54156064d2e81213dc827a689d1

All three of my computers use the same version libXext.so.6.4.0  (verified by md5)
ONE, my home workstation, is the only that does *not* show this error - regardless of which computer I build it - I don't get an error here - but on both the others.   So that pretty much clears the compilation itself. 
Instead there's something in the application that causes the error to be logged some computers.

Could it be that the one working fine have Nvidia proprietary drivers ?  - while the others have Intel ?  (i don't really think drivers have anything todo with this.)

Anyway - judging by the way logs grow, it's clear that only some applications are able to trigger that behavior.
I've tried every usually used application on my laptop, and failed to find even one that generates this error.

billbonney, 2015-01-16:
I propose we close this as a AP2 issue, it's a deployment issue based on platform. We can reopen it if required, to just add dependency to LibXext.so.6.4.0

closed 2015-01-16 by billbonney

AndKe, 2015-01-16:
No, I don't think you need an dependency to that file - just newer version of that very same file that does not complain about the error call it's observing. 
For whoever that googles this issue in hope of solution: copy the file /usr/lib/x86_64-linux-gnu/libXext.so.6.4.0   from a Ubuntu 15.04    - and the issue will not be logged anymore.