ArduPilot/apm_planner #594: Warn if Auto missions plan altitude higher than Fence

R-Lefebvre, 2015-01-30:
We have just fixed an issue in Arducopter that brings back behavior where the vehicle will not be able to climb above the Fence ceiling in Alt_Hold, Loiter, etc.  In these auto-throttle modes, it will not blow through the ceiling and then trigger the fence to RTL, instead, it will stop climbing before the ceiling.

However, we have decided not to enforce this behavior in Auto modes.  This is because it would result in a situation where the the copter would get "stuck", and/or crash into an object the mission was attempting to avoid.

At this point, the Auto mission will run, the copter will go through the ceiling, and then trigger RTL.

It would be great if the GCS systems could look at the state of the Fence_Enabled=1 and Fence_Type=1||3. If true, then check the mission altitudes to make sure that they do not go higher than FENCE_ALT_MAX.  If they do, issue a warning.

billbonney, 2015-01-30:
I've updated the original description to state
>The current behaviour is that the Auto mission will run to completion, and if the copter is above the fence ceiling at the end, an RTL will trigger.

If this is incorrect, please change. Thx

R-Lefebvre, 2015-01-30:
That is incorrect.

Just confirmed in SITL.  It switches to RTL immediately once it passes the fence altitude and flies straight back without finishing the mission.

Unfortunately, we're between a rock and a hard place here, because the user is set two conflicting commands.  There really is not perfect solution other than using the GCS to prevent them taking off like this in the first place.  For sure, if a fence ceiling of 400ft is being used for regulator compliance, this should take precedence over everything else, and this is what it does now.  The only thing we could do differently, is to continue flying the mission at the fenced altitude.  This could result in them crashing into a mountain, etc.  There is no perfect one-size-fits-all solution to this problem.