ArduPilot/apm_planner #595: Preflight calibration: quick accel level missing

hsteinhaus, 2015-02-02:
Old releases of APM planner provided a feature called "preflight calibration". The feature was perfectly suited to quickly adjust accell trim with the vehicle placed on a horizontal surface. Recent releases show a dialog instead, only offering gyro, baro and radio trim calibration.

I used this feature very frequently, as even a slight looser or tighter cable is enough to bring the soft-mounted Pixhawk slightly out of position, causing a drift in Stabilize. The full accell calibration is no replacement for this functionality, as running an accel calib in the field is really a pain with a bulky 10kg octo...

billbonney, 2015-02-02:
Following this discussion https://groups.google.com/d/msg/drones-discuss/4_8FEwxwFqQ/tfy3E8JsRhoJ the option you have are as per Randy's comments.

Which version of FW are you using. If it 3.x selecting Gyro calibration will do the 1D access calibration (i.e. level) you are suggesting.

billbonney, 2015-05-16:
This is related to the new feature #710

closed 2015-05-16 by billbonney