ArduPilot/apm_planner #596: AntennaTracker: Does not forward MAvlink from Radio to USB (or so it seems)

AndKe, 2015-02-02:
This is about using APM 2.5 ,+ AntennaTracker: I did cut UART0 connection to telemetry port, and soldered pads to connect UART2 to Telemetry connector.

Telemetry radio is connected to telemetry port (Serial2), HW flow control disabled on radio, SERIAL2_BAUD = 57, SERIAL2_PROTOCOL = 2
GPS is connected to the usual port.
USB is connected to APM Planner 2

In MissionPlanner 2 , I see the MAVID of the Antenna tracker, but I miss any sign of the UAV (different MAVID)  ,  AHI displayed belongs to AntennaTracker.

Antenna tracker does not turn toward the UAV, but is listed as D|Vector|stabilized, Active, Normal operation. APM Tracker lists is as disarmed, but the servos are holding.

How to troubleshoot ? - right now It does not seem like APM even cares for the incoming serial data, or does it - but fail to forward it ?

The main issue here, is that I see no sign of the UAV.

billbonney, 2015-02-02:
Can you enable the MAVLink Inspector Widget and see which packets you are seeing?

*This is APM Planner 2.0 issue and you wrote Mission Planner 2.0, which do you mean?

AndKe, 2015-02-02:
Hi, sorry for opening this issue, I was so lost.
APM Planner failed to read/set hardware flow control flag from Radio, so I *thought* is was off.  
Once I actually disabled it - I see both MAV's.
Thank you  (the mention of "Mission Planner" , I just verified using it while being desperate)

closed 2015-02-02 by AndKe