ArduPilot/apm_planner #598: User input is not checked propperly in adv parameter list

hsteinhaus, 2015-02-03:
Using latest master, the user is allowed to enter non-numeric values. These are converted to zero internally when the changes are written. 

Just try the following: enter "0,15" on ArduCopter parameter RATE_PIT_P. The value will be accepted, the row changes to green. If you press write now, the value changes into "0.000". This value will be written into the vehicle without further warning.

If your vehicle was in the air while doing this, you better bring your shovel for recovery...

PS.: may be the bug is related to my German locale setting (0,15 in German means 0.15 in English number formatting).

billbonney, 2015-02-03:
I assume this is in the Standard Parameter view? I cannot enter a comma into the edit field. I'll try to set my machine to be german and see if that allows me to enter the data
[image]

hsteinhaus, 2015-02-03:
Ops, sorry for the imprecise wording - I was talking about the "Full Parameter List". But I could imagine that my locale setting (export LANG=de_DE.utf8) is important for this bug.