ArduPilot/apm_planner #599: Multiple MAV_ID (antenna tracker + UAV) makes Primary flight display jump around)

AndKe, 2015-02-03:
When connected to antenna tracker, which forwards UAV data, Primary Flight display is flickering between what looks like the two devices.

Jesse-Millwood, 2016-04-26:
I found this to be an issue also in 2.0.20 with multiple uavs connected. It gets real wild with three. 

[image]

Jesse-Millwood, 2016-04-26:
Also I did make sure everything had either a different source system setting (mavproxy) and all uavs had different SYSID_THISMAV parameters

billbonney, 2016-04-26:
It probably needs to just to filter incoming message based on the ActiveUAS. Likely to be in the layer that links PFD QML to UASObject.

FloodChen, 2018-06-05:
i have the same quetion,do you find any solution?

drobban, 2018-06-07:
@FloodChen what version of apm_planner are you running?

FloodChen, 2018-06-13:
> @drobban apm2.0.24 and apm2.0.18

Arne-W, 2018-06-14:
@FloodChen try 2.0.26-rc2 from here: **[http://firmware.ardupilot.org/Tools/APMPlanner/beta/](http://firmware.ardupilot.org/Tools/APMPlanner/beta/)**

robinhilliard, 2021-05-18:
I'm still seeing this in 2.0.28-rc1