ArduPilot/apm_planner #602: Roi visible in QGCMapWidget

wolkstein, 2015-02-05:
hi,
i make "Set ROI" visible on Map. I am not sure if this make sense for other people. if yes i can create an pull request.
this roi 2d item also work correct with "spline waypoints". see picture.

[image]

/g
wolke

wolkstein, 2015-02-06:
hi,
today i spend a little bit more time to fix some problems on different loiter waypoints.

i noticed that loiter waypoints are displayed without a path. the path only was painted if spline waypoints are used behinf this loiter waypoints.

iattach an image which show some different loiter waypoints painted with the latest apmplanner2.
[image]

after my changes loiter waypoints will painted in this way.
[image]

currently this changes https://github.com/wolkstein/apm_planner/commit/0ba9b74ce89948ab3d31b9b9638e865355f4ea76 are dependent to my first changes on roi. but it is also simply to make one pull request with only this bugfix on loiter waypoints.

/g
wolke

billbonney, 2015-02-06:
You can make one pull request. But in individual commits for easier review. Thx. :+1:

billbonney, 2015-02-06:
Duplicate of #519 and #379

wolkstein, 2015-02-06:
hi,

i create a first pull request https://github.com/diydrones/apm_planner/pull/604 which fix the not drawn path on some waypoints.
second i will create also an pull request for the roi fix.

billbonney, 2015-02-06:
closed with ac4c3cf

commit 428bf690 (2015-02-04): Radio Calibration: Fixes #602 RC_TRIM values not being saved
closed 2015-02-06 by billbonney