ArduPilot/apm_planner #609: APM Planner v2.0.16-rc3

MJBeaubien, 2015-02-10:
I updated my APM Planner this morning and now when I go to Graphs and try to go to Droneshare APM Planner crashes. I've tried numerous times and crashes every time.

billbonney, 2015-02-10:
Thanks, I have fixed, and will be in the next daily build

MJBeaubien, 2015-02-10:
Thanks! I will try it after it's been updated, any idea when that would be?

 

Date: Mon, 9 Feb 2015 17:56:03 -0800

billbonney, 2015-02-10:
about 30mins, you can download a version with the fix here http://firmware.diydrones.com/Tools/APMPlanner/daily/ it will have 2e11c86 in the name

closed 2015-02-10 by billbonney

MJBeaubien, 2015-02-10:
Awesome! That is definitely great to see how fast you dev can fix bugs!

 

Date: Mon, 9 Feb 2015 18:02:08 -0800

MJBeaubien, 2015-02-10:
I don't see it there so far.

 

From: mcbeaubien@hotmail.com
To: reply@reply.github.com
Subject: RE: [apm_planner] APM Planner v2.0.16-rc3 (#609)
Date: Mon, 9 Feb 2015 18:06:31 -0800

Awesome! That is definitely great to see how fast you dev can fix bugs!

 

Date: Mon, 9 Feb 2015 18:02:08 -0800

MJBeaubien, 2015-02-10:
Can I just update my APM Planner?

billbonney, 2015-02-10:
the OSX version is here http://firmware.diydrones.com/Tools/APMPlanner/daily/2015-02-09/apm_planner_2.0.16-rc3-3-g2e11c86_osx.dmg
the windows version is here http://firmware.diydrones.com/Tools/APMPlanner/daily/2015-02-10/apm_planner_2.0.16-rc3-3-g2e11c86_win.exe