ArduPilot/apm_planner #613: Negative values in MAG graphs not represented correctly

andyp1per, 2015-02-11:
In graphs of MAG X,Y,Z values I see values of 65535 etc rather than -1 so someone is decoding a signed short incorrectly. This problem does not appear in Mission Planner 1.3.19, only APM Planner 2.0.15 (Windows and OSX). I'm sure this must be a dup, but cannot find it.

andyp1per, 2015-02-11:
If you need a dataflash log that shows this I can provide it

peterbarker, 2015-07-20:
Is the dataflash log file in question a text log or binary log?

If a text log - how was it produced?  Downloaded, or exported using APM Planner?

andyp1per, 2015-07-20:
Binary, downloaded over mavlink. BTW I stopped using APM Planner for this and other reasons. I occasionally have switched back and it does not seem consistent now - I have seen good MAG graphs, but generally not.

billbonney, 2015-07-20:
Can you post a link to a log, or email me. Thx

andyp1per, 2015-07-20:
https://dl.dropboxusercontent.com/u/108802517/112.BIN
This came directly off the flash card

billbonney, 2015-07-20:
Thanks, I couldn't see any values of 65535 in that log.  But that's with 2.0.18-rc2-17-xxxx see http://firmware.diydrones.com/Tools/APMPlanner/daily/2015-07-16/ 

Can you reproduce the error with the latest build?

peterbarker, 2015-07-20:
Brain dump:

The original binary logs appear correct.  I first found this issue in an exported log, but it appears the problem is in the in-memory model - a problem on log load, as people are having problems with the graphs.

I do not believe this issue reproduces on Linux - I certainly haven't managed it!  What OS are you using, Bill?

It isn't just a boundary problem - I've seen 65530 in (exported text-format) logs.  Definitely seems to be a different-integer-length-being-used issue.

The mav* tools don't cope well with out-of-range values like this in logs.  I'm working up a patch for that.

The relevant load code in AP2DataPlotThread looks correct:
                                else if (typeCode == 'h') //int16_t
                                {
                                    qint16 val;
                                    packetstream >> val;
                                    linetoemit += "," + QString::number(val,'f',0);
                                    valuepairlist.append(QPair<QString,QVariant>(labelstrsplit.at(j),val));
                                }

... bit I think the problem has to lie in there somewhere.

billbonney, 2015-07-20:
I tested this on Xubuntu 15.04 (but at home I run OSX 10.10) I'll test again when I get home.

The graph code was written by @malcom2073 so he knows it best. I'm not so familiar with it. it does seem like sign/unsigned conversion issue if we are seeing 65XXX values.

billbonney, 2015-07-21:
@peterbarker the odd thing about using `QString::number(val,'f',0);` is that val is converted to a double, when you could use `QString::number(uint n, int base = 10)` function instead, for example. I'd expect a int8_t or uint8_t to convert to a uint without a problem in that call.

The loadAsciiLog treats all input as a int on line504

my exported log of 112.BIN loads with errors in 1548 lines  in the log file. which can't be good.

If we had a log with the error reported in loading of a log that would really help.

Anyway, must sleep, I'll look at it more tomorrow if I get some time,.

andyp1per, 2015-07-21:
I'm running on Windows 7 & OSX. I do see the problem more on Windows.

peterbarker, 2015-07-21:
@billbonney Yes, that is odd.  I must admit my eyes skipped over that line on the assumption that the problems were with the in-memory model, and that line was producing some sort of output only.  Perhaps the graph problems were only on loaded text logs?   @andyp1per - could you confirm that the bad graphs are produced when you load a binary dataflash log, please?

andyp1per, 2015-07-21:
All the issues I see are with binary dataflash logs, but that's pretty much all I use so would struggle to prove that this works elsewhere.

billbonney, 2015-07-21:
@andyp1per can you reproduce the problem and let's us know how and with what log, thx

I'm convinced that a an 8bit or 16 bit unsigned to double is going to be wrong conversion.  We should check other places QString::number is being used. I think the conversion to string needs to use its closest type, ie uint, ulong, etc associated function. I just had a similar effect on some other code I was working on. (Ie. Unsigned intro signed, into unsigned) I just like to be able to prove it as well.

andyp1per, 2015-07-21:
Do you have an OSX build?

billbonney, 2015-07-21:
Yep, OSX is my main platform for this.

andyp1per, 2015-07-21:
I don't see it in the link you sent?

billbonney, 2015-07-21:
Sorry, pre :coffee: . Had some now. I haven't implemented a fix yet, but if you can test with the latest here http://firmware.diydrones.com/Tools/APMPlanner/daily/2015-07-18/ and reproduce that would be awesome.

billbonney, 2015-07-21:
this is the changes I am proposing https://github.com/diydrones/apm_planner/tree/logconversion-wip

andyp1per, 2015-07-21:
With the latest pre-release I cannot reproduce this on Windows 7

billbonney, 2015-07-21:
That's good to know. I'm still think the suggested changes make sense, as covering signs can get tricky when the signed numbers are bigger than -32768. You may have experienced an earlier issue for the same reason.

Arne-W, 2017-03-17:
does not happen any more on current master

closed 2017-03-17 by Arne-W