ArduPilot/apm_planner #620: Add Do_Set_Home feature

billbonney, 2015-02-13:
See https://github.com/DroidPlanner/droidplanner/issues/1397 for details

Also need to add warning that once set, it permanent until power cycle
Add feature to update home when arm is received

rmackay9, 2015-04-22:
I'm not sure if this belongs here or in a new issue but..

It is possible for users to take-off without first getting a GPS lock and valid home position. In this case copter will record the home position while the vehicle is flying. This can lead to the home position being quite far from the actual home position or over a location where it's not safe to land.

It would be good if the GCS did this:

 - watch for the home position being set while the vehicle is in flight. The vehicle should send a WP#0 when this happens although the GCS will need to be careful not to confuse this message with a WP#0 received in response to a mission download.
 - if the home position is set during flight the GCS should update the home location on the map and give a verbal warning (optional) to the pilot to say the home has been set and request they move it to a more accurate position
 - allow the pilot to home the home icon around (optionally providing a suggested location using the tablet/computer's GPS) and once set, send a do-set-home command to the vehicle (optionally using the terrain altitude at that horizontal location). Now when the vehicle RTLs it will RTL to the pilot provided home location.

John-Ingersoll, 2016-08-10:
This would also allow better pure Optical Flow missions where the mission plan would be laid out on real coordinates but flown relative to the HOME/launch location.