ArduPilot/apm_planner #623: Camera Gimbal view broken on Mac (v2.0.15)

wanderzell, 2015-02-15:
Initially I thought it was related to old ArduCopter firmware I used, but just upgraded to latest (3.2.something) and the view is broken still.

Basically I have got camera gimbal stabilisation and control working by setting all parameters manually - now it works fine, doing:
- tilt stabilisation
- roll stabilisation
- tilt control via RC6

But when I open Camera Gimbal view all that I see on the attached screenshot. It shows only correctly roll control channel and angle limits, nothing else.

[image]

+ suggestion for usability - why does the screen show "Please set ch7 to camera trigger" in a prominent place with big font? First-time user (as me) thought that is a requirement to get the gimbal working, not that it's just note for guys who use the trigger actually. I'd put it somewhere next to trigger configuration fields, with a slightly smaller text. So that there would be no confusion.

billbonney, 2015-02-15:
What version are you using, since changes have been made to this screen recently? Try 2.0.16-rc3 set beta updates in the config to receive notifications.

billbonney, 2015-02-15:
Fixed with https://github.com/diydrones/apm_planner/commit/cc0e32a4e88cc630ecdeaa86b37deaf70bbac1dd

closed 2015-02-15 by billbonney