ArduPilot/apm_planner #625: Random QtSerial port crashes in apmplanner when opening a serial port.

HarrisonHeli, 2015-02-18:
apmplanner v2.0.16-rc3 on Ubuntu 14.04. QT 5.2.1

I have been getting random SEG faults and application crashes when opening a serial connection. 
It would happen when first loading the application and when reconnecting to a APM 2.6.
It drove me crazy as there was no repetitive or clear message in the logs as to what was going on.

I debugged the application in Qt Creator and could not trigger the failure when single stepping through the code. When running in release mode, the SEG fault would come back.

I ended up putting a 1 sec delay in "/src/comm/serialconnection.cc" to give Qt time to open the port correctly before attempting any r/w access.
I also improved the debug comments in the portError section.

https://github.com/HarrisonHeli/apm_planner/commit/41d346a9c02e654f443423607026d9c5faa153b2

So far the problem has stayed away. Anyone else having this issue?

HarrisonHeli, 2015-02-19:
Back again.
I believe I have located the source of the SEG Fault.

When valid data is detected on the serial port and when the UASManager object has determined what type of UAS it has connected, Two key steps are carried out in the UASManager::addUAS() function.

Step 1) emit UASCreated -> caught by MainWindow::UASCreated()
MainWindow::UASCreated creates  all the UI objects for that type of UAS (Plane,Rover,Quad). One of them happens to be PrimaryFlightDisplayQML

Step 2) Calls UASManager::setActiveUAS()
This emits ActiveUASSet message - caught by UASManager::ActiveUASSet
Within ActiveUASSet(), data is added to the PrimaryFlighDisplayQML UI 

If Step 1 happens before Step 2, then everything runs ok.

If Step 2 is carried out before Step 1 because the UASCreated message was delayed in getting to the MainWindow, I get a SEG Fault.

Any Ideas on implementing the correct solution. 
And Thanks to Bill Bonney for giving me some guidance.

billbonney, 2015-02-19:
It should be easy enough to show that in a log using Qlog statements, or add a QAssert to break when the object are not initialized,

HarrisonHeli, 2015-02-20:
Ok I have fixed it.

The member variables of class AbsPositionOverview, RelPositionOverview and  VehicleOverview are not being initialised to known values in the class constructors.

Normally this would be OK as there would be some sort of checking to see if the data is valid before using the variables. This would be done within the class functions or by the calling class.

The problem is PrimaryFlightDisplay.qml requests the values form the above classes and assumes that value will be valid and within range. 
So when it gets a random pitch value like 536377252, it triggers a SEG Fault.

I am going to call the issue closed and do a bit more testing before submitting.

closed 2015-02-20 by HarrisonHeli