ArduPilot/apm_planner #628: Terminal outputs strange characters, won't take input

jaredmoody, 2015-02-21:
I'm on os x 10.10.2, and I am making sure to connect within 30s of connecting power to the board. 

[image]

madfish01, 2015-02-21:
ArduCopter 3.2-rc1 9-May-2014
Changes from 3.1.4
 11) CLI removed from APM1/2 to save flash space, critical functions moved to MAVLink:

https://github.com/diydrones/ardupilot/blob/master/ArduCopter/ReleaseNotes.txt
http://diydrones.com/forum/topics/cli-giving-garbage-with-apm-2-6-with-3-2-firmware

billbonney, 2015-02-21:
Thanks @madfish01 issue closed

closed 2015-02-21 by billbonney

jaredmoody, 2015-02-21:
Ah, makes sense, thanks guys.