ArduPilot/apm_planner #629: UDP connection parameters are forgotten across app restarts

chris1seto, 2015-03-02:
With the pixhawk streaming packets over UDP, I start the APM Planner and add a new UDP connection. I specify a port to listen, and then hit the "Add IP" button to set the IP and port number of the remove device. The connection immediately becomes active and I can both read and write to the Pixhawk.

If I restart the APM Planner, MAV link packets are still received and the UI reflects this, but I can no longer write mavlink packets to the device until I go back into the options for the UDP connection and re-enter the IP and port in the "Add IP" dialog.

Not working on both .15 and 16 RC3.

chris1seto, 2015-03-02:
Here's the log after setting up the connection, closing APM Planner, and reopening it. http://pastebin.com/7j3VCHhR

billbonney, 2015-03-02:
Have you tried deleting the ~/.config/com.diydrones/ folder with the settings in it? Do this after you have closed APM Planner

chris1seto, 2015-03-02:
I have not been able to locate this file on windows. It's not in
    either the apmplanner2 dir in my user account of in the hidden
    folder AppData. Any idea of it's location?
    
      
    
    On 3/2/2015 11:11 AM, Bill Bonney
      wrote:
    
    
      Have you tried deleting the ~/.config/com.diydrones/ folder
        with the settings in it? Do this after you have closed APM
        Planner
      —
          it on GitHub.

chris1seto, 2015-03-02:
I found the .ini file, deleted it, and tried again. No change.

chris1seto, 2015-03-02:
Here is my config file after setting up the link. It clearly saves the link correctly, but perhaps is having some issue reopening it after the app restarts?  http://pastebin.com/UziRqGaH