ArduPilot/apm_planner #631: Droneshare Upload Failure: Invalid location data was found in that log, ignoring

RedPhoton, 2015-03-04:
I'm seeing this pretty consistently on APM Planner v2.0.15 on the Mac with logs from a Pixhawk. (I'd be happy to include my dataflashLog but don't  know how -- I'm new). I've plotted lat/lon and the values look ok. Is there something more I can do/provide?  Thanks!

[image]

malcom2073, 2015-03-04:
Can you upload the log file to dropbox, or the ardupilot forums? That is a reply back from the droneshare service, so likely not an AP2 issue... but I'll verify.

RedPhoton, 2015-03-05:
Thanks! The log file is too big for the ardupilot forms (greater than 2MB). Here is a dropbox link, https://www.dropbox.com/l/rBSFlKf8Yy0onHI5h5AyNq

malcom2073, 2015-03-05:
Looks like it's a case of https://github.com/diydrones/droneshare/issues/146 I exported the bin to log, removed the ERR line, and it uploaded just fine. I recommend posting your log there, letting them know it's an issue. I've created an issue to make the error messages AP2 provides more informative: https://github.com/diydrones/apm_planner/issues/634

closed 2015-03-05 by malcom2073

visser, 2015-05-01:
It's still happening as of latest Mission Planner on Windows, I went into the .log file and removed the ERR line (when battery failsafe kicked in) at the end of the file then uploaded the log manually to DroneShare.