ArduPilot/apm_planner #638: Full parameter list completely broken in 2.0.16 and beyond

hsteinhaus, 2015-03-12:
Commit 6cd6f596ad688e62c1be72e0e1a06b7398e96db4 breaks the full parameter list. It is no longer possible to change values. All values are rejected with message "Failed to convert number, please verify your input and try again", regardless of their type or validty. After reverting the commit 6cd6f, everything works again.

My system: Linux 64bit, QT 5.4.1, German locale setting.

teamosceola, 2015-03-12:
I don't have this problem. I'm running the latest git version of arducopter and apm_planner.

My system: Ubuntu 14.04.2 LTS 64bit

billbonney, 2015-03-13:
@teamosceola did you try to enter a number in German format I.e 5.6 would be 5,6

hsteinhaus, 2015-03-13:
@billbonney 
The problem is not directly related to floating point format. Even trying to change integers (like ARMING_CHECK) triggers the error.

malcom2073, 2015-03-13:
@hsteinhaus Can you paste the output of "locale" here? I've set my locale to de_DE.UTF-8, and do not see the issue, but pasting the output of that command would help to match my environment to yours.

Also what distro/distro-version?

hsteinhaus, 2015-03-13:
This is Centos 7. Just changing the LANG variable (that is enough to revert most programs to plain English locale) has no influence on the error.

Here is the output  of locale

LANG=de_DE.UTF-8
LC_CTYPE="de_DE.UTF-8"
LC_NUMERIC="de_DE.UTF-8"
LC_TIME="de_DE.UTF-8"
LC_COLLATE="de_DE.UTF-8"
LC_MONETARY="de_DE.UTF-8"
LC_MESSAGES="de_DE.UTF-8"
LC_PAPER="de_DE.UTF-8"
LC_NAME="de_DE.UTF-8"
LC_ADDRESS="de_DE.UTF-8"
LC_TELEPHONE="de_DE.UTF-8"
LC_MEASUREMENT="de_DE.UTF-8"
LC_IDENTIFICATION="de_DE.UTF-8"
LC_ALL=

malcom2073, 2015-03-13:
There's the key. I was unable to duplicate the issue on Ubuntu, but just tried it on Fedora 21 and it has the issue, I'll look into it, thanks for the report!

malcom2073, 2015-03-30:
@hsteinhaus can you try the lastest master? I believe I fixed this as of 08ca5a890c72243a850f2890c47b840f3d94ff2a so 5,6 should work with german locale, and 5.6 with english.

hsteinhaus, 2015-03-30:
Partial success ;-). APM Planner now accepts integer or valid german fp numbers (e.g. 1,01) in the full param list.

Unfortunately all numbers in the table that were read from the drone are in english notation (1.01). So if you try to change a gain from 1.00000001 to 1.00000002, you have to delete the whole number and re-enter it in it's german notation (or change the '.' for a ','). Not really good usability IMHO.

BTW, I think when using a program with an english UI, no german user would ever expect having to enter german numbers (there are so many non-localized programs out there...). For my perception it would be completely ok to force the user into the english number formatting and leave all locale stuff away, as long as no full localization (i.e. translated menus) is provided.

malcom2073, 2015-03-30:
How about accepting both, but only displaying it with a point rather than a comma? At least until we can localize the whole thing.

hsteinhaus, 2015-03-30:
Should probably work for 99% of the users, but still contains chances for ambiguities. The dot is used for digit grouping here, similar as the comma in the english locale (mainly in the business context, but who knows....)

All what I have learned about localizing in nearly 20 years of software engineerig: do it (very!) consequently or leave it. Anything between is only causing a million of bugs. And doing it consequently is a damn hard job, not just initializing a few variables.

BTW, the best example for that thesis that I have ever seen is Mission Planner. It has causes two very expensive holes in my flying field that way...

hsteinhaus, 2015-04-02:
Just found out that the fix breaks compatibility to all existing param files (they contain english fp numbers). When loading one of these files, apm planners goes into a (nearly) endless loop of error messages (Failed to convert...), one for every row.