ArduPilot/apm_planner #639: Notify user of removal of APM1/APM2 support for AC3.3

malcom2073, 2015-03-13:
As APM1/APM2 are no longer supported as of AC3.3, AP2 should notify users when they attempt to install AC3.3 on an APM1/APM2, rather than just throwing an error about being unable to find the firmware on the server.

billbonney, 2015-03-16:
The user should just be presented with the available latest version. I.e 3.2.1 for APM:Copter on APM2.x HW I don't think they should be 'warned' . Seems odd. ;-)

zlite, 2015-03-16:
Agreed. I thought the dev team had already talked about sending them to the
last known good version for APM. At any rate,  let's do that.

malcom2073, 2015-03-17:
Added support for "trunk" in 512a779e30cb528b7e717b949cc67831e715ebff, leaving this issue open until beta and eventually stable also are put in.