ArduPilot/apm_planner #643: 3axis gyro calibration does not work on current arduplane

AndKe, 2015-03-16:
I did not test stable arduplane.
Arduplane built from master: an attempt to do 3axis calibration shows a 30 sec countdown, but no instructions to change attitude.

Even older mission planner works with the Pixhawk and same firmware, which leads me to think the problem is not limited to current master arduplane only.

martinjuju, 2015-06-09:
same problem with arducopter 3.3rc5

AndKe, 2015-06-09:
That's very strange.
The report is old, and should be closed.
Just yesterday, I did a 3-axis gyro stabilisation on 3.3rc5 , a week ago on arduplane3.3.
What version on APM Planner do you use ?

billbonney, 2015-06-09:
This was fixed in 2.0.18-rc1, please retest, you can reopen if it's still a problem

closed 2015-06-09 by billbonney