ArduPilot/apm_planner #644: segmentation fault after mavlink connect

AndKe, 2015-03-17:
This kind of fault happens now and then, right after connection.

https://gist.github.com/AndKe/f2737c5ce8a60ea0c87e

malcom2073, 2015-03-17:
can you type "bt" after it crashes to get a stack trace?

AndKe, 2015-03-17:
here is a new crash:  propely logged.
Please tell me if some +DEBUG should be used during compilation, for more symbols or anything :) 
https://gist.github.com/AndKe/2b37799c7b4fc32118e3

malcom2073, 2015-03-17:
AndKe: can you try out the branch michael_dev? (af4cc61d547e3e8b6f46d20cf39daf059869edb7) I pulled in https://github.com/diydrones/apm_planner/pull/627 which may fix it.

AndKe, 2015-03-17:
It works very well, no crash.
tried a lot of connects /restart om planner+connects , but no crash.
I did experience a connect-but-no-data issue like https://github.com/diydrones/apm_planner/issues/645 - but no crash.

Thanks.

malcom2073, 2015-03-30:
Closing this as fixed in (5a763084857e1306a8b4fa43a0d87ed2aab0c48d)

closed 2015-03-30 by malcom2073