ArduPilot/apm_planner #645: Sometimes, I connect - but are ubable to get parameters.

AndKe, 2015-03-17:
I suspect this is related to another issue that happens now-and-then;  connecting, but not seeing AHI or status updates.

This log shows a connection, that is partially working - I see staus, AHI, ++ , but no paramters are received.
https://gist.github.com/AndKe/f455053516f37ad42a3f

lostcontrol, 2015-03-28:
I'm new to Ardupilot and APM planner (actually new to the all multicopter thing) and I'm seeing the same issue (2.0.16-15-g8ff7139 compiled from source on Ubuntu 14.10) with my Pixhawk clone.

malcom2073, 2015-03-30:
@lostcontrol Can you give c9718ee099af4ea718414f9fc3a104d9474d9933 a shot and see if it works now? It should now check every 10 seconds to make sure the parameter request went through, and if not, resend it to avoid this.

lostcontrol, 2015-03-30:
@malcom2073 Unfortunately, it does not help. But I think the problem is unrelated to APM Planner. I found this bug report:

https://github.com/diydrones/ardupilot/issues/1161#issuecomment-77726759

And since I already had some issues with USB 3.0 ports with another device (Logitech Harmony), I suspected this could be the problem. But disabling USB 3.0 in the BIOS does not seem to help.

The strange thing is that I was able to connect successfully several time in a row without any issues earlier today. Now, I can't manage to connect anymore since more than one hour and many Pixhawk restarts and laptopt reboots...

The point is that connecting using *screen /dev/ttyACM0 57600 8N1* works not reliably. The connection gets stuck at some point. E.g.

```Using external compass[0]

Press ENTER 3 times to start interactive setup

�3�Calibrating barometer_R```

And then nothing more. The LEDs on the Pixhawk look good. So the issue is either with my Pixhawk or some USB related problem on my laptop.

Just in case, here is the log with c9718ee099a: https://gist.github.com/lostcontrol/0bbb797e8fa7b70e33d9

AndKe, 2015-03-30:
sounds like a good solution, unfortunaltly due to this vessels firewalled
satlink, I am getting a useless text-only github page that says:

 Param Manager: Added timer to check if UAV missed the parameter reque…

…st message, and resend

   - Loading branch information...

 commit c9718ee099af4ea718414f9fc3a104d9474d9933 1 parent 8ff7139
<https://github.com/diydrones/apm_planner/commit/8ff71391f51eeb387839d3d0acd29bdab2cdcdf9>

so I am unable to checkout the proper (which) branch , even git pull does
work (slowly)

AndKe, 2015-03-30:
I can communicate by mail only; but if it helps; I have this problem on
Dell Latitude E5520 , no USB3.0 on it.

lostcontrol, 2015-03-30:
I did a quick test with another laptop also running Ubuntu 14.10 and connecting/disconnecting multiple times with *screen /dev/ttyACM0 57600 8N1* works (I get some garbage characters at regular interval on the terminal). I have not APM Planner on this laptop.

I rebooted once more my laptop and did the same test again. Seemed to work this time. I fired APM Planner and it connected successfully. I unplugged/plugged the Pixhawk a few times and was able to connect successfully every time.

So I don't have any good explanation for this problem but I don't think it is related to APM Planner at all (at least in my case) and rather to my hardware and/or Linux drivers.

AndKe, 2017-03-17:
havent seen it since ~ 2015

closed 2017-03-17 by AndKe