ArduPilot/apm_planner #646: CPU hogging after connect,

AndKe, 2015-03-18:
APM planner takes about 30% CPU time on a quad core 2.4Ghz laptop ftom the moment of serial connection.
Disconnecting makes it drop less then 5% , to about 26%
This goes also for non-graphic views whithout maps/AHI

malcom2073, 2015-03-30:
Relatively high cpu usage is expected, but it should not be continuing after disconnection. I am able to reproduce this on Windows, so I'll try to figure out what's spinning while disconnected.

AndKe, 2015-06-06:
Confimed by another user here: https://github.com/diydrones/apm_planner/issues/716

AndKe, 2015-09-12:
how did this issue get cloased ?
It's still hogging CPU like crazy, and extended laptop battery can't help that.
Starts after telemetry connection.

AndKe, 2015-09-12:
To reproduce:
-connect to pixhawk using USB or radio.
-observe the *application* occupy for example 15% CPU within 1 munite of connection
-disconnect , remove USB radio/Pixhawk , observe that CPU use continues (by apmplanner2)to climb
- vait 10 minutes, , you may easily be at 35% cpu by then...

please reopen..

AndKe, 2015-09-12:
Additional info: "Primary flight display" is most likely to blame.
Reproducable:
If I allow the CPU hogging  problems to start, closing "primary flight display" widget does not help. (maybe a clue here!?)
If I close "primary flight display" before connecting/changing tabs, then the problem will not appear.  
Also: "primary flligt display (old)" does not cause this.
Finally: just using "primary flight display" does not cause the problem by itself, changing tabs and then back to flight data tab where it is does.

AndKe, 2015-09-13:
Final additional info:
The GCS laptop is a Latitude 5520 , i7 CPU , Ubuntu 15.04 x64 , with intel graphics(CPU)
"normally" working APM Planner2 is consuming about 3% CPU, a little more than playing a HD video.

my workstation has a faster CPU, and Nvidia GPU - usual apmplanner2 CPU usage is <1% , and the problem is *not* reproducible on this system.

So we could blame it on the Intel graphics, except I have not found any  ohter app behaving this way, and it's caused by this one widget. 

Also, tab switching is trigging the problem, but also minimizing and restoring the apmplanner2 window.

billbonney, 2015-09-15:
I'll see if I can reproduce this on my linux machine.

Q: If you reconnect to the serial device, to another serial device, does the CPU usage return to normal?
Q: Have you tried updating the graphics drivers
Q: Have you tested to see if this is a issue on Qt5.4.2 or Qt5.5?

AndKe, 2015-09-15:
Q1; not sure if I understood that right, but I can disconnect by click from tha APM/Radio, or even physically remove the device from USB bus, (tty* device disappears too) - and the CPU will remain loaded as before.    

Q2; Tested now on two processors, i5-2430M and i7-4770 (both used for graphics) , same issue- but less prominent on the faster CPU, as I understand xserver-xorg-video-intel uses open Intel code.
I too would suspect driver problem, but the fact that the issue starts only during a connection, contradicts it.

Q3;  Sorry, Qt 5.5 is not regarded as stable for this distro yet, will be there eventually, https://wiki.ubuntu.com/Touch/QtTesting

AndKe, 2015-09-20:
Intel have started to provide driver-installer for ubuntu 15.04 yesterday.
Installing Intel graphics drivers did not help at all, there is no change in behaviour.
- This collaborates the fact that a graphics error should be there all the time, not only after telemetry connection.
Also, this issue may be related to https://github.com/diydrones/apm_planner/issues/801  - it does not help to unload "primary flight display" widget - which makes me believe that there is a chance that things, or too many things keep running.

billbonney, 2015-09-20:
"Q3; Sorry, Qt 5.5 is not regarded as stable for this distro yet, will be there eventually,"
Which version of Qt then? (sorry it's unclear to me)

AndKe, 2015-09-20:
andre@Loke:~$ qmake -v
QMake version 3.0
Using Qt version 5.4.1 in /usr/lib/x86_64-linux-gnu

AndKe, 2016-04-09:
After checking with Qt 5.5.1 - and narrowing down the issue , I changed the title.
reproduced on ubuntu running on a Dell Latitude E5520 (intel graphics)

more findings:
1:  after starting AP2 and minimizing it, then transmitting some UDP telemetry to it,  - CPU hogging starts even if it was minimized at all times (NO pixel drawed) 
2: if I start AP2 , disable the "primary flight display" widget, then close AP2  - then start AP2 - let it see some telemetry , stop the mavlink stream, then enable "primary flight display"    - the CPU hogging will start 

number 2 tells me that there's something wrong with it drawing anything else than the default, level, AHI, because it does not hog CPU on loading before data, but only one dataset to be displayed is enough for it to start.
- Also.;  the "primary flight display (old)"  widget never hogs CPU.

AndKe, 2016-05-24:
**# IMPORTANT UPDATE:**

the easiest way to reproduce:
-open "apmplanner2"  click "graphs", "enable log playback" "replay logfile" - and select any .tlog 
let it play, or move to a spit where artificial horizon changes (primary flight display widget in "flight data screen") 
- observe CPU hogging. 
-now you can stop playback, or even close the widget.
- observe that CPU is still hogged...

Arne-W, 2017-03-21:
I did some testing and can confirm this behaviour - especially that this "hogging" only happens on intel systems. My laptop which have intel graphics show this issue, my Workstation with nvidia does not.
Closing the widget will not help as the object won't be destructed - it just gets invisible.
It will be very tricky to find the culprit cause it seems somehow hardware/driver related.

Using the old primary display seems to be a usable workaround on intel graphics - are there any drawbacks using the old one?

Arne-W, 2017-03-21:
Okay I think I found a solution :smiley: 
@AndKe could you please test [THIS](https://github.com/Arne-W/apm_planner/tree/test-Anti-Hog) branch?
It is **VERY** important that you start the executable from a shell and set " QSG_RENDER_LOOP=threaded" in the environment (by typing "export QSG_RENDER_LOOP=threaded" into the shell before starting the apmplanner)
For me it works! Only drawback of the solution is the environment variable which must be set. This cannot be done from the code - as far as I know.

[Here](https://forum.qt.io/topic/68721/high-cpu-usage) is some further information about this issue.

Arne-W, 2017-03-21:
I have tested this change with Intel and Nvidia Graphic cards. On Nvidia it does not change anything and on intel it solves the problem. It would be great if someone could test this change on an ATI graphics card - just to be sure it has no impact there.

AndKe, 2017-03-21:
Fantastic,  - it works, I chose to replay a longer mission with some 70 WP's as a long test, and it was very exciting the eqivalent of see one core being busy while the WP's got received, but once done, apmplanner dropped to the usual ~5% CPU   .    The need for enviroment variable could be solved by loading it by an shellscript, instead of the execuatable.     ....I do not have an AMD GPU to test with.

Arne-W, 2017-03-22:
Very good! As I have not a clue about the installation stuff someone should adapt it to install such a script on linux systems. Anybody out there who can do this?

AndKe, 2017-03-22:
it's very likely I can figure it out, - if the enviroment variable
absolutely necessary from the start, or can it be added once the fix is in
master ?

Arne-W, 2017-03-22:
No - it seems like its the other way round. Using only the environmental setting without the code change is supposed to cause problems. It shouldn't be a problem to bring it in without the script.
I will make a PR

AndKe, 2017-03-22:
apmplanner2.sh  with ```
#!/bin/bash
#variable required to prevent CPU hogging in Intel video chipsets.
export QSG_RENDER_LOOP=threaded
./apmplanner2
```

should be used to execute it then.