ArduPilot/apm_planner #648: Random crash (assertion & core dump) xcb_conn.c, queue_len failed

ZacVawter, 2015-03-18:
The crash seems very random in nature, sometimes it runs stable for 30 minutes, sometimes it barly starts before crashing.  I've tried 2.0.15 installed from the .deb, and compiled locally.  Running Ubuntu 14.04.  I'm running connected via a UDP proxy running on the local host.

Console log:
--------------------------------------------------------------------------------------
 INFO 2015-03-18T15:23:46.832 QGCCore::initialize() 
 INFO 2015-03-18T15:23:46.833 Current Build Info 
 INFO 2015-03-18T15:23:46.833 Git Hash: 08ca5a890c72243a850f2890c47b840f3d94ff2a 
 INFO 2015-03-18T15:23:46.833 Git Commit: 2.0.16-6-g08ca5a8 
 INFO 2015-03-18T15:23:46.833 APPLICATION_NAME: "APM Planner" 
 INFO 2015-03-18T15:23:46.833 APPLICATION_VERSION: "v2.0.16" 
 INFO 2015-03-18T15:23:46.833 APP_PLATFORM: ubuntu64 
 INFO 2015-03-18T15:23:46.833 APP_TYPE: stable 
 INFO 2015-03-18T15:23:46.851 Start Link Manager 
DEBUG 2015-03-18T15:23:46.851 Create MAVLinkDecoder:  MAVLinkDecoder(0x29d4900) 
 INFO 2015-03-18T15:23:46.852 Start UAS Manager 
DEBUG 2015-03-18T15:23:46.873 Creating MainWindow 
 INFO 2015-03-18T15:23:46.938 Creating SubMainWindow:  SubMainWindow(0x2aa5c30) 
 INFO 2015-03-18T15:23:46.938 setting objectName:  "VIEW_MISSION" 
 INFO 2015-03-18T15:23:46.954 Creating SubMainWindow:  SubMainWindow(0x2b2f1f0) 
 INFO 2015-03-18T15:23:46.954 setting objectName:  "VIEW_FLIGHT" 
 INFO 2015-03-18T15:23:46.965 Creating SubMainWindow:  SubMainWindow(0x2b4c7a0) 
 INFO 2015-03-18T15:23:46.965 setting objectName:  "VIEW_HARDWARE_CONFIG" 
DEBUG 2015-03-18T15:23:47.033 fillPortsInfo  
 INFO 2015-03-18T15:23:47.036 Inserting  "ttyUSB0" 
 INFO 2015-03-18T15:23:47.036 Inserting  "ttyUSB1" 
 INFO 2015-03-18T15:23:47.036 Inserting  "ttyACM0" 
 INFO 2015-03-18T15:23:47.036 Inserting  "ttyS0" 
 INFO 2015-03-18T15:23:47.036 Changed Link to: "ttyUSB0" 
 INFO 2015-03-18T15:23:47.209 Creating SubMainWindow:  SubMainWindow(0x2ba6d20) 
 INFO 2015-03-18T15:23:47.210 setting objectName:  "VIEW_SOFTWARE_CONFIG" 
DEBUG 2015-03-18T15:23:47.222 Disable Controls 
 INFO 2015-03-18T15:23:47.340 Creating SubMainWindow:  SubMainWindow(0x39ce830) 
 INFO 2015-03-18T15:23:47.340 setting objectName:  "VIEW_ENGINEER" 
QObject::connect: No such signal QTableView::currentCellChanged(int,int,int,int)
QObject::connect:  (sender name:   'tableWidget')
QObject::connect:  (receiver name: 'AP2DataPlot2D')
 INFO 2015-03-18T15:23:47.357 Creating SubMainWindow:  SubMainWindow(0x3a59610) 
 INFO 2015-03-18T15:23:47.357 setting objectName:  "VIEW_SIMULATOR" 
DEBUG 2015-03-18T15:23:47.378 MODE: 0 
DEBUG 2015-03-18T15:23:47.378 "D|PREFLIGHT" 
DEBUG 2015-03-18T15:23:47.378 MODE: 80 
DEBUG 2015-03-18T15:23:47.379 "D|STABILIZED" 
DEBUG 2015-03-18T15:23:47.379 MODE: 64 
DEBUG 2015-03-18T15:23:47.379 "D|MANUAL" 
DEBUG 2015-03-18T15:23:47.379 MODE: 88 
DEBUG 2015-03-18T15:23:47.379 "D|VECTOR|STABILIZED" 
DEBUG 2015-03-18T15:23:47.379 MODE: 92 
DEBUG 2015-03-18T15:23:47.379 "D|AUTO|STABILIZED" 
DEBUG 2015-03-18T15:23:47.380 MODE: 66 
DEBUG 2015-03-18T15:23:47.380 "D|TEST" 
DEBUG 2015-03-18T15:23:47.428  QUrl( "file:///home/randy/apm_planner/apm_planner/release/qml/PrimaryFlightDisplayQML.qml" )  
DEBUG 2015-03-18T15:23:47.491 QML Status: 1 
DEBUG 2015-03-18T15:23:47.492  QUrl( "file:///home/randy/apm_planner/apm_planner/release/qml/PrimaryFlightDisplayQML.qml" )  
DEBUG 2015-03-18T15:23:47.532 QML Status: 1 
 INFO 2015-03-18T15:23:47.535 UASActionsWidget creating  UASActionsWidget(0x402a960) 
DEBUG 2015-03-18T15:23:47.537 QuickView: load settings 
DEBUG 2015-03-18T15:23:47.538 QuickView: loading key:  "GCS Status.Altitude (GPS) (m)"  type: "text" 
DEBUG 2015-03-18T15:23:47.540 QuickView: loading key:  "GCS Status.Altitude (REL) (m)"  type: "text" 
DEBUG 2015-03-18T15:23:47.541 QuickView: loading key:  "GCS Status.Battery (%)"  type: "text" 
DEBUG 2015-03-18T15:23:47.542 QuickView: loading key:  "GCS Status.Climb (m/s)"  type: "text" 
DEBUG 2015-03-18T15:23:47.544 QuickView: loading key:  "GCS Status.Current (A)"  type: "text" 
DEBUG 2015-03-18T15:23:47.545 QuickView: loading key:  "GCS Status.GPS Fix ()"  type: "text" 
DEBUG 2015-03-18T15:23:47.547 QuickView: loading key:  "GCS Status.GPS HDOP (m)"  type: "text" 
DEBUG 2015-03-18T15:23:47.549 QuickView: loading key:  "GCS Status.GPS Sats ()"  type: "text" 
DEBUG 2015-03-18T15:23:47.551 QuickView: loading key:  "GCS Status.Pitch (deg)"  type: "text" 
DEBUG 2015-03-18T15:23:47.554 QuickView: loading key:  "GCS Status.Roll (deg)"  type: "text" 
DEBUG 2015-03-18T15:23:47.557 QuickView: loading key:  "GCS Status.Voltage (V)"  type: "text" 
DEBUG 2015-03-18T15:23:47.559 QuickView: loading key:  "GCS Status.Yaw (deg)"  type: "text" 
DEBUG 2015-03-18T15:23:47.564 QuickView: load settings END 
 INFO 2015-03-18T15:23:47.601 Using Alsa Audio driver 
[... 599 lines cut ...]
DEBUG 2015-03-18T15:24:11.838 Param: "BATT_MONITOR" :  QVariant(int, 0) 
DEBUG 2015-03-18T15:24:11.863 Received BATT_VOLT_PIN combo index to: 13 
DEBUG 2015-03-18T15:24:11.863 Param: "BATT_VOLT_PIN" :  QVariant(int, 13) 
DEBUG 2015-03-18T15:24:11.884 Received paramter BATT_CURR_PIN 
DEBUG 2015-03-18T15:24:11.884 Param: "BATT_CURR_PIN" :  QVariant(int, 12) 
DEBUG 2015-03-18T15:24:11.907 Received BATT_VOLT_MULT parameter 
DEBUG 2015-03-18T15:24:11.907 Param: "BATT_VOLT_MULT" :  QVariant(double, 10.1) 
DEBUG 2015-03-18T15:24:11.927 Received BATT_AMP_PERVOLT parameter 
DEBUG 2015-03-18T15:24:11.928 Param: "BATT_AMP_PERVOLT" :  QVariant(double, 17) 
DEBUG 2015-03-18T15:24:11.951 Param: "BATT_AMP_OFFSET" :  QVariant(double, 0) 
DEBUG 2015-03-18T15:24:11.972 Received BATT_CAPACITY "3300" 
DEBUG 2015-03-18T15:24:11.973 Param: "BATT_CAPACITY" :  QVariant(int, 3300) 
DEBUG 2015-03-18T15:24:11.996 Param: "BATT2_MONITOR" :  QVariant(int, 0) 
DEBUG 2015-03-18T15:24:12.018 Param: "BATT2_VOLT_PIN" :  QVariant(int, 13) 
DEBUG 2015-03-18T15:24:12.040 Param: "BATT2_CURR_PIN" :  QVariant(int, 12) 
DEBUG 2015-03-18T15:24:12.063 Param: "BATT2_VOLT_MULT" :  QVariant(double, 10.1) 
DEBUG 2015-03-18T15:24:12.086 Param: "BATT2_AMP_PERVOL" :  QVariant(double, 17) 
DEBUG 2015-03-18T15:24:12.108 Param: "BATT2_AMP_OFFSET" :  QVariant(double, 0) 
DEBUG 2015-03-18T15:24:12.131 Param: "BATT2_CAPACITY" :  QVariant(int, 3300) 
DEBUG 2015-03-18T15:24:12.154 Param: "MIS_TOTAL" :  QVariant(int, 0) 
DEBUG 2015-03-18T15:24:12.176 Param: "MIS_RESTART" :  QVariant(int, 0) 
DEBUG 2015-03-18T15:24:12.199 Param: "RALLY_TOTAL" :  QVariant(int, 0) 
DEBUG 2015-03-18T15:24:12.222 Param: "RALLY_LIMIT_KM" :  QVariant(double, 5) 
apmplanner2: ../../src/xcb_conn.c:186: write_vec: Assertion `!c->out.queue_len' failed.
Aborted (core dumped)

malcom2073, 2015-03-30:
I believe this is related to this fix (https://github.com/diydrones/apm_planner/pull/627) which was pulled in. Can you verify it still happens on the latest master build? 

Or, can you get me a stack trace? After compiling from source:

$ gdb apmplanner2
(Once GDB loads)
$ r
(Once it crashes)
$ bt

projectkk2glider, 2015-04-24:
Same error on Linux Mint 17.1 amd64 using commit e76141cc65cd1030978d7ed9e72b9f088f684319

The crash happens at random times. I am using Bluetooth for connection to APM.

Backtrace:
```
QFont::setPointSizeF: Point size <= 0 (-0.950000), must be greater than 0
QFont::setPointSizeF: Point size <= 0 (-0.950000), must be greater than 0
QFont::setPointSizeF: Point size <= 0 (-0.950000), must be greater than 0
apmplanner2: ../../src/xcb_conn.c:186: write_vec: Assertion `!c->out.queue_len' failed.

Program received signal SIGABRT, Aborted.
[Switching to Thread 0x7fff7b7fe700 (LWP 8756)]
0x00007ffff1e4ecc9 in __GI_raise (sig=sig@entry=6) at ../nptl/sysdeps/unix/sysv/linux/raise.c:56
56	../nptl/sysdeps/unix/sysv/linux/raise.c: No such file or directory.
(gdb) bt
#0  0x00007ffff1e4ecc9 in __GI_raise (sig=sig@entry=6) at ../nptl/sysdeps/unix/sysv/linux/raise.c:56
#1  0x00007ffff1e520d8 in __GI_abort () at abort.c:89
#2  0x00007ffff1e47b86 in __assert_fail_base (fmt=0x7ffff1f98830 "%s%s%s:%u: %s%sAssertion `%s' failed.\n%n", 
    assertion=assertion@entry=0x7fffea1d2615 "!c->out.queue_len", file=file@entry=0x7fffea1d2600 "../../src/xcb_conn.c", line=line@entry=186, 
    function=function@entry=0x7fffea1d2630 "write_vec") at assert.c:92
#3  0x00007ffff1e47c32 in __GI___assert_fail (assertion=0x7fffea1d2615 "!c->out.queue_len", file=0x7fffea1d2600 "../../src/xcb_conn.c", line=186, 
    function=0x7fffea1d2630 "write_vec") at assert.c:101
#4  0x00007fffea1c6e5b in ?? () from /usr/lib/x86_64-linux-gnu/libxcb.so.1
#5  0x00007fffea1c7191 in ?? () from /usr/lib/x86_64-linux-gnu/libxcb.so.1
#6  0x00007fffea1c78a7 in ?? () from /usr/lib/x86_64-linux-gnu/libxcb.so.1
#7  0x00007fffea1c79c8 in xcb_take_socket () from /usr/lib/x86_64-linux-gnu/libxcb.so.1
#8  0x00007ffff0571a38 in ?? () from /usr/lib/x86_64-linux-gnu/libX11.so.6
#9  0x00007ffff0572259 in _XFlush () from /usr/lib/x86_64-linux-gnu/libX11.so.6
#10 0x00007ffff0574c75 in _XGetRequest () from /usr/lib/x86_64-linux-gnu/libX11.so.6
#11 0x00007ffff056e0bb in XSync () from /usr/lib/x86_64-linux-gnu/libX11.so.6
#12 0x00007ffff2a3f75a in ?? () from /usr/lib/x86_64-linux-gnu/mesa/libGL.so.1
#13 0x00007ffff341bed1 in QOpenGLContext::swapBuffers(QSurface*) () from /usr/lib/x86_64-linux-gnu/libQt5Gui.so.5
#14 0x00007ffff5e32470 in ?? () from /usr/lib/x86_64-linux-gnu/libQt5Quick.so.5
#15 0x00007ffff5e32c08 in ?? () from /usr/lib/x86_64-linux-gnu/libQt5Quick.so.5
#16 0x00007ffff2cf6233 in ?? () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#17 0x00007ffff0ec8182 in start_thread (arg=0x7fff7b7fe700) at pthread_create.c:312
#18 0x00007ffff1f1247d in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:111

```

I appears to be a bug in libxcb. Mine is:
```
$ dpkg -l | grep libxcb
ii  libxcb-composite0:amd64                     1.10-2ubuntu1                                       amd64        X C Binding, composite extension
ii  libxcb-dri2-0:amd64                         1.10-2ubuntu1                                       amd64        X C Binding, dri2 extension
ii  libxcb-dri2-0:i386                          1.10-2ubuntu1                                       i386         X C Binding, dri2 extension
ii  libxcb-dri2-0-dev:amd64                     1.10-2ubuntu1                                       amd64        X C Binding, dri2 extension, development files
ii  libxcb-dri3-0:amd64                         1.10-2ubuntu1                                       amd64        X C Binding, dri3 extension
ii  libxcb-dri3-0:i386                          1.10-2ubuntu1                                       i386         X C Binding, dri3 extension
ii  libxcb-dri3-dev:amd64                       1.10-2ubuntu1                                       amd64        X C Binding, dri3 extension, development files
ii  libxcb-glx0:amd64                           1.10-2ubuntu1                                       amd64        X C Binding, glx extension
ii  libxcb-glx0:i386                            1.10-2ubuntu1                                       i386         X C Binding, glx extension
ii  libxcb-glx0-dev:amd64                       1.10-2ubuntu1                                       amd64        X C Binding, glx extension, development files
ii  libxcb-icccm4:amd64                         0.4.1-1ubuntu1                                      amd64        utility libraries for X C Binding -- icccm
ii  libxcb-image0:amd64                         0.3.9-1ubuntu2                                      amd64        utility libraries for X C Binding -- image
ii  libxcb-keysyms1:amd64                       0.3.9-1ubuntu1                                      amd64        utility libraries for X C Binding -- keysyms
ii  libxcb-present-dev:amd64                    1.10-2ubuntu1                                       amd64        X C Binding, present extension, development files
ii  libxcb-present0:amd64                       1.10-2ubuntu1                                       amd64        X C Binding, present extension
ii  libxcb-present0:i386                        1.10-2ubuntu1                                       i386         X C Binding, present extension
ii  libxcb-randr0:amd64                         1.10-2ubuntu1                                       amd64        X C Binding, randr extension
ii  libxcb-randr0-dev:amd64                     1.10-2ubuntu1                                       amd64        X C Binding, randr extension, development files
[... cut ...]
ii  libxcb-xv0:amd64                            1.10-2ubuntu1                                       amd64        X C Binding, xv extension
ii  libxcb1:amd64                               1.10-2ubuntu1                                       amd64        X C Binding
ii  libxcb1:i386                                1.10-2ubuntu1                                       i386         X C Binding
ii  libxcb1-dev:amd64                           1.10-2ubuntu1                                       amd64        X C Binding, development files
```

projectkk2glider, 2015-04-24:
It is a libxcb bug indeed, Ubuntu 14.04 is affected. Same problem in other projects https://github.com/Swordfish90/cool-retro-term/issues/21  and https://github.com/glfw/glfw/issues/386

Possible solution https://gist.github.com/slimsag/a26d838ccc4480ce21bc

Arne-W, 2017-03-20:
According to [ https://bugs.launchpad.net/ubuntu/+source/libxcb/+bug/1402966](https://bugs.launchpad.net/ubuntu/+source/libxcb/+bug/1402966) the xcb bug is fixed

closed 2017-03-20 by Arne-W