ArduPilot/apm_planner #650: Log Viewer Message Selection Glitch

labtoast, 2015-03-19:
The graph view has a bug where the message type selection view is not scrolling properly. I think the height of the viewable area is not being calculated correctly when message types are expanded. The scrolling does work correctly when no message types are expanded.

Steps to reproduce:
1. Launch apm_planner
2. Click graphs mode
3. Load log file
4. Expand a few message types
5. Arrow keys can now navigate to out of view message types.

[image]
[image]
[image]
[image]
[image]

malcom2073, 2015-03-30:
Which version of AP2 are you running? Help->About APM Planner 2.0, the first two lines of text.

labtoast, 2015-04-02:
From "Help->About APM Planner 2.0":
APM Planner v2.0.16
(8ff71391-2.0.16-15-g8ff7139)

From codebase:
commit 8ff71391f51eeb387839d3d0acd29bdab2cdcdf9

System:
Qt 5.3.2
OSX 10.10.2

It looks like I can scroll further off the viewable area by expanding additional message types.

malcom2073, 2015-04-02:
Can you do a full rebuild (clean, qmake, build)? It looks like you missed some UI changes, which can happen if you don't do a fresh build after a checkout.

The telltale is the message selection dialog is now the full height of the screen, making the table view less wide. This is what it looks like now:

[image]

Arne-W, 2017-03-17:
seems to be fixed

closed 2017-03-17 by Arne-W