ArduPilot/apm_planner #654: .bin log not parsed correctly, type code 'h' treated as uint16_t

Smartype, 2015-04-01:
I am try to add a mouse based optical flow sensor (ADNS 5050) to my setup. When debugging I found Dx, Dy is displayed as unsigned value. found the bug in `apm_planner/src/ui/AP2DataPlotThread.cc`

```
diff --git a/src/ui/AP2DataPlotThread.cc b/src/ui/AP2DataPlotThread.cc
index b75f51b..1bcf5fe 100644
--- a/src/ui/AP2DataPlotThread.cc
+++ b/src/ui/AP2DataPlotThread.cc
@@ -205,7 +205,7 @@ void AP2DataPlotThread::loadBinaryLog()
                                 }
                                 else if (typeCode == 'h') //int16_t
                                 {
-                                    quint16 val;
+                                    qint16 val;
                                     packetstream >> val;
                                     linetoemit += "," + QString::number(val,'f',0);
                                     valuepairlist.append(QPair<QString,QVariant>(labelstrsplit.at(
```

I should have created a pull request, but the link speed to github is so slow here. Cloning the repo takes too much time. ;-)

Thanks!

billbonney, 2015-04-01:
We can make the change. Look at using the --depth parameter to limit the amount of history your retrieve. i.e. --depth=5

Smartype, 2015-04-01:
Good idea! Thanks.

commit b0819775 (2015-04-01): Graph View: Fix #654, logs loaded using unsigned when they should used signed for 'h' values
closed 2015-04-01 by malcom2073