ArduPilot/apm_planner #657: Maps are in Portuguese, home locked to old 3DR office location

zlite, 2015-04-08:
This issue report has been confirmed!

http://ardupilot.com/forum/viewtopic.php?t=11903&p=31694#p31694

malcom2073, 2015-04-09:
https://github.com/diydrones/apm_planner/blob/master/libs/opmapcontrol/src/core/opmaps.cpp#L42

Looks like it defaults to Portuguese... whups! I'll default it back to English, and add a TODO that it should be handled by localization.

Unsure as of yet about the home location being locked, looking into that now.

billbonney, 2015-04-09:
I'm not sure that's the cause, (unless it's been around a long time) since I think it maybe related to the google maps api update https://github.com/diydrones/apm_planner/commit/a8f7d91b6d8d83d99f629ece8c21274b27179bd0#diff-c6daef03e42938db59f4530a79c192d9

silvermanphoto, 2015-04-09:
> I think it maybe related to the google maps api update a8f7d91
> #diff-c6daef03e42938db59f4530a79c192d9

Thanks, Bill.  I admire your hard work on this.  APM Planner is great and
getting better with each release.

I wonder if you would let me get involved in improving the Wiki?
Documentation of the Parameters and Log types needs to get fleshed out;  a
new user has to do a forum search for every unknown and hope that it's been
asked about.  If a user wonders, say, whether it's better to
set WP_YAW_BEHAVIOR to "face along GPS course" instead of pointing to
the next waypoint, you have to dig really deep.  A good editor could go
fast through the Param list in one or two sessions in a Mumble voice call
with you, Randy, etc. and turn that conversation into expanded Wiki
documentation with an efficient and limited set of follow-up emails.

I previously offered this help to TCIII, and he said there is already an
editor, but in six months of daily user involvement, I haven't seen any
Wiki progress.  Will you please let me help?  I'm a trained researcher,
writer and teacher with 25 years of experience doing this.

Best,
Joel Silverman

malcom2073, 2015-04-09:
Perhaps the API just started paying attention to the language being "pl-PL"? Looks like we've been sending that, but getting English until recently.

billbonney, 2015-04-10:
@silvermanphoto send me an email, you'll find it from my github profile

commit 7c09afe4 (2015-04-10): Maps Widget: Fixes #657 map shown in Portuguese

Note: will need to delete ~/apmplanner2/mapscache dir
closed 2015-04-10 by billbonney