ArduPilot/apm_planner #658: Add support for DO_DIGICAM_* messages in mission planning screen

jmachuca77, 2015-04-12:
Currently only DO_DIGICAM_CONTROL (ID #203) mission items are supported in the mission planning sceen, but you cannot set any of the fields for the message. Please add the fields and description to the control message, and also add support for DO_DIGICAM_CONFIGURE (ID #202). 

DO_DIGICAM_CONFIGURE fields:
Mode | Shutter Speed | Aperture | ISO | ExposureMode | CommandID | Engine Cut-Off

DO_DIGICAM_CONTROL fields:
On/Off | Zoom Position | Zoom Step | Focus Lock | Shutter Cmd | CommandID

billbonney, 2015-04-12:
How does that correlate with this description. http://copter.ardupilot.com/wiki/mission-planning-and-analysis/mission-command-list/#do-digicam-control

jmachuca77, 2015-04-13:
The DO_DIGICAM messages have more to them than is currently on ardupilot. There is a change coming that will add a parameter for camera_type and if the camera supports the other functions. The wiki will be updated once the change is tested. The ground stations need to support the official MAVLINK command structure as defined here:

https://pixhawk.ethz.ch/mavlink/

billbonney, 2015-04-13:
@jmachuca77 Can you specify more clearly the behaviour based on the parameter, the parameter name etc. If we showed all the parameters always, it just creates confusion for users when they try and set something they cannot. Most  fields shown in the mission items are only subsets of the MAVLink protocol. Limited to what is implemented in ArduPilot.

It's the reason why DO_DIGICAM_CONTROL is limited to 'trigger shutter' action at the moment.

closed 2018-08-16 by jmachuca77