ArduPilot/apm_planner #659: No way to load previous Firmware versions

mboland, 2015-04-13:
In the initial setup window/Install Firmware tab there is now way to load previous versions of the firmware.
This need has happened with issues in 3.2.1 and I needed to go back to the more stable 3.1.5 version.
There is a custom firmware option but that requires you to manually download a previous version and then select the file on your drive. Just finding a previous version in firmware/diydrones can keep you looking for days.

malcom2073, 2015-04-15:
As you said, finding a previous version can keep you looking. There's no actual list of previous versions that is kept anywhere along with binaries. The issue has been raised with the Ardupilot team, and if they implement some form or archive for previous versions it can be implemented as a list to select from in AP2.

mboland, 2015-04-15:
Now that the versions of firmware have different hardware supported it is becoming more important to be able to choose what firmware version you need. 
Previously all efforts were put into getting the bugs ironed out, but now we have 3.2.1 which has problems but the developer focus is on 3.3, rather than ironing out 3.2
Also, Mission Planner has a previous FW list in the FW upgrade window. How is Michael doing this if the developers are not supporting it?

proficnc, 2015-04-16:
Firmware.diydrones.com

billbonney, 2015-04-16:
@proficnc yes that is where all are them are, now which one in AC3.2.1 release

this one is stable http://firmware.diydrones.com/Copter/stable/PX4-quad/git-version.txt
now post me the links to 3.2 release version?

mboland, 2015-04-16:
And the 3.1.5 release for APM2.6 Hex please.

billbonney, 2015-04-16:
here's the issue I have raised with ardupilot to provide an official list. see https://github.com/diydrones/ardupilot/issues/2118

@meee1 maintains his own list, which is OK, but it really should be part of the build & release process for  ardupilot