ArduPilot/apm_planner #661: Crash during tlog playback.

AndKe, 2015-04-15:
Once the mission is loaded , APM planner will crash.
to reproduce: download https://drive.google.com/a/aranica.com/file/d/0B3E4K1nvpDuFVkMtUHc1YlF3WUE/view?usp=sharing

And playback at 10x speed.

malcom2073, 2015-04-15:
Assuming you're on the latest master as of your post?

AndKe, 2015-04-15:
Yes, and the segmentation fault will occour soon after it reads WP10  - the file works with MP, and this is not the only tlog that gives this result.

malcom2073, 2015-04-15:
I do not see this on Windows, I can play it back at 1x or 10x without issue. Can you get a stack trace?

AndKe, 2015-04-15:
I can reproduce it every time,  regardless of playback speed, now it crashed after WP13:
'''
DEBUG 2015-04-15T17:04:50.989 UPDATING WAYPOINT 12 IN 2D MAP 
DEBUG 2015-04-15T17:04:50.989 src/ui/map/QGCMapWidget.cc 736 UPDATING WP FUNCTION CALLED 
DEBUG 2015-04-15T17:04:50.989 UPDATING WAYPOINT 13 IN 2D MAP 

Program received signal SIGSEGV, Segmentation fault.
0x00000000007333c9 in WaypointEditableView::setCurrent(bool) ()
(gdb) bt
#0  0x00000000007333c9 in WaypointEditableView::setCurrent(bool) ()
#1  0x000000000064bb57 in WaypointList::currentWaypointEditableChanged(unsigned short) ()
#2  0x000000000064bc23 in WaypointList::currentWaypointViewOnlyChanged(unsigned short) ()
#3  0x00000000009d0aac in WaypointList::qt_static_metacall(QObject*, QMetaObject::Call, int, void**) ()
#4  0x00007ffff322012a in QObject::event(QEvent*) () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#5  0x00007ffff4d6f537 in QWidget::event(QEvent*) () from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#6  0x00007ffff4d3116c in QApplicationPrivate::notify_helper(QObject*, QEvent*) () from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#7  0x00007ffff4d3641e in QApplication::notify(QObject*, QEvent*) () from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#8  0x00007ffff31f081b in QCoreApplication::notifyInternal(QObject*, QEvent*) () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#9  0x00007ffff31f27e3 in QCoreApplicationPrivate::sendPostedEvents(QObject*, int, QThreadData*) () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#10 0x00007ffff3247a63 in ?? () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#11 0x00007fffec167c5d in g_main_context_dispatch () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#12 0x00007fffec167f48 in ?? () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#13 0x00007fffec167ffc in g_main_context_iteration () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#14 0x00007ffff3247e57 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) ()
   from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#15 0x00007ffff31ee042 in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#16 0x00007ffff31f5acc in QCoreApplication::exec() () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#17 0x000000000043dbb1 in main ()
(gdb) 
'''

malcom2073, 2015-04-15:
Can you recompile with debugging enabled?
$ make clean
$ qmake CONFIG+=DEBUG qgroundcontrol.pro
$ make
And then run?

AndKe, 2015-04-15:
I did that now,  looks good, bu the output /(stacktrace looks just as it was last time ((above)

andre@Loke:~/prog/apm_planner$ qmake CONFIG+=DEBUG qgroundcontrol.pro
Project MESSAGE: Qt version 5.3.0
Project MESSAGE: Linux build x64_86
Project MESSAGE: Release flavor
Project MESSAGE: BASEDIR /home/andre/prog/apm_planner DESTDIR /home/andre/prog/apm_planner/release TARGET apmplanner2
Project MESSAGE: Skipping support for QUpgrade
Project MESSAGE: Using mavlink  + ardupilotmega
Project MESSAGE: Skipping support for OpenSceneGraph
Project MESSAGE: Skipping support for Google Earth
Project MESSAGE: Skipping support for Protocol Buffers
Project MESSAGE: Skipping support for libfreenect
Project MESSAGE: Adding ALGLIB /home/andre/prog/apm_planner/libs/alglib
Project MESSAGE: Skipping support for XBee
Project MESSAGE: Skipping support for Opal-RT
Project MESSAGE: QTDIR /usr
andre@Loke:~/prog/apm_planner$

Arne-W, 2016-02-20:
Made a fix....

AndKe, 2016-02-22:
great !  - thank you, I am surprised over the complexity.

Arne-W, 2016-02-22:
complexity - hmm just 2 sanity > 0 checks....

billbonney, 2016-02-22:
That commit https://github.com/Arne-W/apm_planner/commit/5313ee572fc7afa0bfd66d6f49ab3d9873df71d9 looks strange to me. I think it should read !=0 since using the < implies it can be negative, which it can't as it's an unsigned int 

That waypoint point is awful since it as a 'protocol sized' bytes in the definition which it should have, it should really just be an integer not a quint16

Arne-W, 2016-02-23:
For me the code `if (seq > 0)` only implies that seq must be greater than zero. Sure in this case it can't be negative cause seq is unsigned so `if (seq != 0)` would do the same job. In my opinion the `if (seq > 0)` is more universal as it works with signed and unsigned types.
But there is another issue I think - something with the first seq must go wrong... If seq must be greater than zero this line `if (seq < waypoints.count())` will make a problem.
I will check that - perhaps this fix was too simple ;-)

Arne-W, 2016-02-25:
Next approach :-)
Last one was a real bad one - but this one is perfect :-) cause it fixes the bug and does not make a workaround.

commit 19c59aa1 (2016-03-05): Merge pull request #910 from Arne-W/pull_NewFix_661

Graph View: BugFix for crash while replaying tlog - Fixes #661
closed 2016-03-05 by billbonney