ArduPilot/apm_planner #662: Please reduce zoom sensitivity in maps.

AndKe, 2015-04-15:
Using a 2-finger touchpad scroll on a map is very hard due to to very high sensitivity. 
Please reduce it to about 1/6

malcom2073, 2015-04-15:
Does a mouse wheel also scroll it very fast, or just two finger touchpad scrolling?

AndKe, 2015-04-15:
Wheel sensitivity is fine (and stock speed)  - it works great in a browser, a spreadsheet, and for scrolling thru the log , or zooming in graphs in APM planner.
but the maps have very limited amount of zoom levels, and the scrolling is very fast.  
Due to map tile loading (or even over cellular network) - and need to maintain cursor at area of interest while zooming, a fine zoom is needed, or else it's easy to get lost :)

malcom2073, 2015-04-15:
The reason I ask, is that both mouse wheel and touchpad scrolls are handled identically. Decreasing speed to suit a touchpad would make the mouse wheel very slow. I'll look into it and see if there is any way to detect which is used. If the graph works ok, there should be some way.

billbonney, 2015-04-15:
On what platform? Since OSX/Windows & Linux all behave differently with mouse wheel controls. I think a better solution is to use double click to zoom, or to use only onscreen zoom buttons

AndKe, 2015-04-15:
Linux (Ubuntu)
You may be right, also, https://github.com/diydrones/apm_planner/issues/663
will reduce needed amount of panning/scrolling.   (While reviewing mission
plan and the UAV is booting)

closed 2015-09-18 by AndKe