ArduPilot/apm_planner #667: APM Planner won't calibrate accelerometers APMCopter 3.3 rc1

rbachtell, 2015-04-16:
Accelerometer calibration times out and will not initialize pixhawk under 3.3rc1.  Revert back to 3.2.1 and works normally.  I tried 4 different pixhawks with the same results, two of them new out of the box.  Using latest stable release of APM Planner.

billbonney, 2015-04-26:
I have just tested this against AC3.3RC1 and it works as expected using APM Planner 2.0.17 master.

You can try the latest daily build from http://firmware.diydrones.com/Tools/APMPlanner/daily/2015-04-26/
To see if that fixes  your issue.

PS: Please raise support type discussions here http://ardupilot.com/forum/viewforum.php?f=82 first. If it a bug we can raise it in github then. 

Thanks for reporting all the same :+1:

closed 2015-04-26 by billbonney

rbachtell, 2015-04-26:
Thanks Bill,  I just finished a hex build and tried to calibrate accels with no results, read your email and loaded the latest build which worked just fine.  Thanks for your help and for helping to develop an outstanding GCS.

Randy