ArduPilot/apm_planner #668: when .tlog is is open as log (not played back) - modes are not visualized.

AndKe, 2015-04-16:
"Show mode" checkbox is green, HEARTBEAT.custom_mode can be graphed, but is not decoded into readable markers.

AndKe, 2015-04-16:
It may be related, but roll, pitch, yaw and maybe other values cannot be graphed properly, some appear as integers only , some are just "0"   - same tlog show proper values when loaded for playback.

malcom2073, 2015-04-17:
Correct, tlogs don't have a MODE variable, they use custom_mode. We need a way to know what kind of vehicle a tlog is replaying for in order to correctly display custom_mode.

roll/pitch/yaw however should not be 0, I'll look at that.

billbonney, 2015-04-17:
The heartbeat message contains the vehicle type

Arne-W, 2016-03-06:
Made a fix :-)

commit 8e4cc974 (2016-03-06): Graph View: Mode text arrows now work for loaded Tlogs too. Fixes #668
closed 2016-03-06 by billbonney