ArduPilot/apm_planner #673: Allow changing "Info view" order of items/colors.

AndKe, 2015-04-17:
The Info View widget is nice and useful. but in order to place items, all needs to be unchecked, the checked in the correct order.   - Then they will appear in planned position, but with random colors, often barely readable on white chackground.

please allow color and position setting, and maybe widget background color setting.

malcom2073, 2015-04-17:
If you double click on an item, you can change what is in that particular position. It would be nice if colors could be swapped, as well as a better interface for switching.

AndKe, 2015-04-17:
Thanks, now I discovered that the order is not saved anyway, I set Voltage & Current on the lower "row" , and after resatarting APM planner, Voltage and Yaw is on the lower line.