ArduPilot/apm_planner #675: Request: Ctrl-S should save changed values

AndKe, 2015-04-19:
Whenever in full parameter list, or in an tuning windows with save button, please ass "Ctrl-S" shortcut to save values, or maybe Ctrl-w to "write" to UAV ? 
- reducing mouse movement and time when tuning. - and removing the need to focus out of the parameter being adjusted.

billbonney, 2015-04-19:
This is as easy fix, just go to the .ui file and add keyboard shortcut to the button. See the ArduPlanePidConfig.ui for example

AndKe, 2015-04-19:
I will be out on a mission next few days, but hope to have a pull request ready on Tuesday on Wednesday.
Thank you.

AndKe, 2015-04-22:
please accept my pull requests :)

billbonney, 2015-04-22:
I will. Sorry been busy.

billbonney, 2015-04-26:
These PRs have been pulled in, Thx

closed 2015-04-26 by billbonney