ArduPilot/apm_planner #676: APM Planner 2.0.15 crash on Open

lvale, 2015-04-20:
Just simply trying to open APM Planner 2.0.15 on MacBook Pro late 2013 (dual GPU's) and sometimes it crashes 

A link for a crash report is here:
https://www.dropbox.com/s/mgbex4c8si13on1/apm%20planner%202015%20crash%20on%20open.txt?dl=0

billbonney, 2015-04-20:
Please retest with 2.0.17, if you still have issues, feel free to reopen. thx

lvale, 2015-04-20:
Sorry. Issue occurred with 2.0.17 not 2.0.15.

My mistake.

billbonney, 2015-04-26:
I'd suggest downloading and re-installing. You will then re-affrim gatekeeper settings. I am not seeing this as an issue with 2.0.17 downloaded from the internet.

(I did see that 10.10.3 did update gatekeeper settings so that maybe the issue)

lvale, 2015-05-02:
Apparently the OS X 10.10.3 update might be the cause, as you say, because the updates gatekeeper settings.
Re installed 2.0.17 and started APM Planner using the right mouse button.
[image]

closed 2015-05-02 by lvale