ArduPilot/apm_planner #685: AC3.3 renames some parameters which affects the Extended Tuning screen

rmackay9, 2015-04-24:
The Loiter PID and Throttle params have changed.  I'll have to provide more details although it is in an email somewhere on drones-discuss.

billbonney, 2015-05-24:
see https://groups.google.com/forum/#!searchin/drones-discuss/VEL_XY_P/drones-discuss/hysXUlKDs8Q/15f0t6QJ9cEJ

billbonney, 2015-05-25:
Here's a first draft layout, please comment on what you would like to be where. I will adjust the layout a bit as it's not quite right.
[image]

lthall, 2015-05-25:
Hi Bill,
Could you please add the filter frequency to the Rate Loops.
The other parameters that users may want to tune are:

ATC_ACCEL_P_MAX
ATC_ACCEL_R_MAX
ATC_ACCEL_Y_MAX
MOT_CURR_MAX
MOT_THST_BAT_MAX
MOT_THST_BAT_MIN
MOT_THST_EXPO
MOT_THST_MAX

If it helps, when I tune a copter I go through the standard calibration and setup. I then do an autotune where I am looking at these parameters:
AUTOTUNE_AGGR
AUTOTUNE_AXES
Generally I will do an autotune on roll and pitch with AGGR set to 0.1. Then I will do Yaw. Then I will set AGGR to 0.05 and do it again and see if my noise levels are low enough for autotune to work with 0.05.

Depending on my results from the first autotune I may adjust these parameters:
INS_GYRO_FILTER,20
RATE_PIT_FILT_HZ,20
RATE_RLL_FILT_HZ,20
generally if I am having problems with autotune I need to increase them to 40.

So as you can see, during the whole tuning process the variables I manually change are:
AUTOTUNE_AGGR
AUTOTUNE_AXES
INS_GYRO_FILTER
RATE_PIT_FILT_HZ
RATE_RLL_FILT_HZ
and the channel 7/8 option to enable autotune.
The rest of the parameters above I don't change but look at to see what autotune did.

Thanks!!

billbonney, 2015-05-26:
Thanks,

If you have a preference in how the layout is for the screen, feel free to draw you preferred version. It's easy to add and move around the spin boxes and labels. It's harder to know what the work flow should be. (just draw it on paper and take a picture if that works for you)

rmackay9, 2015-05-26:
Here are a couple of layout options
[image]
[image]

billbonney, 2015-05-31:
Closing this issue with commit 503e689. Have new Issue #720 to track rework of UI layout

closed 2015-05-31 by billbonney