ArduPilot/apm_planner #686: Segfault trying to connect via TCP

BurtMacklin, 2015-04-25:
After entering IP and port and trying to connect, apm_planner crashes on me.

Observed this in

* the current release 2.x version on a Windows 7 machine
* a build from a 02/2015 pulled source on Debian

billbonney, 2015-04-25:
Am I correct in assuming you have built this version yourself from code? If so it seems that you need to clean your build. I have not problem using TCP/UDP connections on my linux build on (linux Mint 17) 14.10 (32bit) or Xubuntu 15.04 (64bit).

JackLaLanne, 2015-04-25:
Yes, for the Linux test, I built from an unknown master revision from some point in February.

Just checked the pre-compiled windows version and that was 2.0.14. I am now using the auto updated to test with the 2.0.14 release.

billbonney, 2015-04-25:
Fo your own build you need to be trying 2.0.17 at a minimum ( I don't have an issue on linux with that build)
What do you mean by 'current release' please post a log and version number thanks. Current Release is 2.0.17, is that the version you are running on Win7?

JackLaLanne, 2015-04-25:
I wrote I was updating the windows to 2.0.14 while really I meant 2.0.17.
Will get to try that out tomorrow and report back.

JackLaLanne, 2015-04-27:
Ok, TCP link between ebay APM 2.5.1 and APM Planner 2.0.17 (47bd731d-2.0.17) on Windows 7 64. This is going through GSM/3G on both plane and gs via openvpn. Here are my findings and thoughts:

I did it like that: Check file/advanced mode, communication/add link/TCP, enter ip and port, connect

*no program crash (bug fixed)
*link seems to be established since green box flashes (heartbeat) and constantly updated sensor data is displayed
*there is no real feedback whether the connection is established, connect window just disappears
*going to communication and selecting my TCP link again, connection window reopens and indeed reveals: connected
 *it seems like the TCP connections are not selectable from the menu on the top right (there is just MAV0) and the connect button/icon there falsely suggest a disconnected status.
*parameter download status is not displayed when connecting
*yet full parameter list shows all parameters. refreshing parameters works flawlessly, here download status is displayed

billbonney, 2015-04-28:
I just fixed the issue of the active link not being displayed. Try a latest daily build from http://firmware.diydrones.com

closed 2015-04-28 by billbonney

JackLaLanne, 2015-04-28:
Just tested as above with

apm_planner_2.0.17-36-g249bc8e_win.exe	2015-04-28 01:15 	130 M

and the connect status works well, now making the overall process smoother than using mission planner.

Thanks!

[image]