ArduPilot/apm_planner #689: Remote radio settings are not read. (3DR telemetry)

AndKe, 2015-04-26:
After reading local radio settings, Ststus on remote radio changes to "Read radio eeprom values"  - there is no sigh of actual readout.

billbonney, 2015-04-26:
This works as expected for me. What version of the SiK FW do you have loaded? Do the radios connect 'solid green light'?
if you are linux you can use screen to connect
`screen /dev/<device_name> 57600'
if you type +++ there time and pause you enter command mode
does typing 'RTI' return value?

AndKe, 2015-04-26:
Hi. 
Yes before reporting - I made sure both radios worked both ways using mission Planner in VMware.
This is direct output from Screen.
ATI
SiK 1.9FFN on HM-TRP
RTI
SiK 1.9FFN on HM-TRP
(the firmware is modified for encryption, but it's SIK 1.9 based, and 100% command compatible with SiK)

FYI: both radios were connected to computer - so they were idle at the time of testing. - I could do the same test with one connected to a Pixhawk if we have any reason to believe that should be needed. (which I don't)

uiliw, 2015-05-20:
I got the same error when i try to copy to remote, both radio get solid green but cant send any data.. Using APM Planner v2.0.17

[image]

billbonney, 2015-05-20:
The reason for this issue posted here is that this maybe related to the SiK 1.9FFN FW.
A quick test for the SiK1.9 FW works as expected.
Support Request are handled on the ardupilot.com forum, see http://ardupilot.com/forum/viewforum.php?f=82
[image]

AndKe, 2015-05-20:
@billbonney  please note that @uiliw uses the very same firmware.
Also, I told you that there were no changes in standard SiK AT commands in my firmware.  
Can we start guessing other possibilities, like, can it somehow be OS related ? - I am using Linux, what did you (both) test with ?

uiliw, 2015-05-20:
@billbonney ill try update the firmware and see if works! thanks for the forum link!
@AndKe im using OSX Yosemite v10.10.2

AndKe, 2015-05-20:
@uiliw  you posted a screenshot of SiK 1.9 - that's the current version. (nothing to update)

billbonney, 2015-05-20:
@AndKe I'm not saying there isn't an issue with SiK1.9FFN, I just haven't had chance to test it yet. if you post link to the FW it will make it easier for me to test.
I can test linux today (I tested on OSX before)

uiliw, 2015-05-20:
hehe, yeap, i just realized that!! So i definitely dont know what to do.

Att.
Willy Werlang

Interface Solutions & Design
Web & Multimedia UI Design
HTML & Graphic Design

www.uili.com.br
E-mail: contato@uili.com.br
MSN: wwerlang@hotmail.com
Skype: dj_uili
Fone: (48) 91675952

AndKe, 2015-05-20:
I tried to test on APM Planner for windows - https://github.com/diydrones/apm_planner/issues/712
Within minutes, I'll test two standard SiK 1.9 radios ASAP.

AndKe, 2015-05-20:
Mea culpa.
Yes, encryption firmware triggers the problem - Strangely ,it works fine in MissionPlanner as you can see from the screenshot, there must be some character, or maybe just the length of the name that causes something to fail when getting remote data, will look into it later.

billbonney, 2015-05-21:
Here's a screen shot of it working on Xubuntu 15.04. The issue with the encryption fw is that it probbaly sends some extra parameters  with the ATI5 call, so it breaks the state machine.
[image]

makuser, 2015-09-13:
Any update on this? The bug still exists in v2.0.18@Ubuntu15.04.
ATI and RTI both return "SiK 1.9 on HM-TRP", so the connection must be established properly.

AlivaDot, 2017-03-15:
After change baudrate back to 57600 - Read radio eeprom values is dissapied

Arne-W, 2017-03-17:
This issue should be fixed through #1004.
@AlivaDot what do you mean exactly?