ArduPilot/apm_planner #697: Mission elevation widget:  does not show anything in red, or add elevation.

AndKe, 2015-04-28:
Debug messages shows that it does get altitudes from Google just fine,  - but it does not seem to "use" in any way.

DEBUG 2015-04-28T20:23:11.460 elevation loc:  69.6311 , 18.7309 
DEBUG 2015-04-28T20:23:11.460 elevation alt 4 : 0.242471 
DEBUG 2015-04-28T20:23:11.460 elevation loc:  69.6256 , 18.7133 
DEBUG 2015-04-28T20:23:11.461 elevation alt 5 : 21.6489 
DEBUG 2015-04-28T20:23:11.461 elevation loc:  69.6201 , 18.6957 
DEBUG 2015-04-28T20:23:11.461 elevation alt 6 : 75.2947 
DEBUG 2015-04-28T20:23:11.462 elevation loc:  69.6145 , 18.6782 
DEBUG 2015-04-28T20:23:11.462 elevation alt 7 : 122.357

billbonney, 2015-05-06:
this is fixed with 10076e22d8ff85cba63c57cbb6e9dcb6a6d7af28

closed 2015-05-06 by billbonney

AndKe, 2015-05-07:
Somehow, current master build, with this fix , does not graph anything (not even blue) in mission elevation widget, nor do I see any attempt to get the elevation in the console/log, or any error.

billbonney, 2015-05-07:
I just tested with a loaded mission and I ahve this. You might need to be connected to a drone?
[image]

AndKe, 2015-05-08:
Yes, now I see that it works fine when connected.
The mission does not need to  be on the UAV, it works with mission editor data.
Could you please make it work whithout a UAV connected ?

I guess MissionElevationDisplay::activeUASSet(UASInterface *uas) needs to be called from the end of MissionElevationDisplay::MissionElevationDisplay(QWidget *parent)    - but I am not sure about proper parameters.   

As a planning tool, we should not need an active connection for this widget to work.

billbonney, 2015-05-09:
What you say makes sense. It requires using
`UASWaypointManager *UASManager::getActiveUASWaypointManager()` API to get a Waypoint manger before a UAS is connected.

One you have connected and disconnected you can still edit the waypoints, it's just that first connection creates a UAS object which hold the UASWaypointManager for the UAS.

AndKe, 2015-05-09:
How can it be done ? Can something be called from MissionElevationDisplay::MissionElevationDisplay(QWidget *parent) , ? , or somewhere else ?