ArduPilot/apm_planner #698: Refreshing tlogs - not all shown

JackLaLanne, 2015-05-01:
Tried to download flash logs via 3G today. Refreshing the logs present on the device, only a few files were display.

Without proceeding to download or remove any logs, I reconnected via USB serial and found twice the number of logs.

billbonney, 2015-05-01:
Can you provide a log from ~/apmplanner2 directory

JackLaLanne, 2015-05-03:
I just attempted to reconstruct the problem in order to provide a log.

Now the refreshing via 3g works really fast and the same number of log files is shown as when connected by usb.

Maybe last time 3g link was too laggy.

JackLaLanne, 2015-05-05:
Tried really hard to make the problem come up again but still all was fine. If it reappears, I will paste the log file.
During the process I noticed a little snag with the log download window. Once logs appear to have finished refreshing and you click done, nothing happens.