ArduPilot/apm_planner #701: Maps caching

lvale, 2015-05-02:
While caching a larger area for offline usage APM Planner caches several zoom levels, which might take a long time....

While the time it takes is not "controlled" by APM Planner, one setting on APM Planner is kind of annoying, and could be changed.

Currently, after a zoom level is cached, APM Planner stops and asks if the following zoom level is to be cached and just sits there.....This dialog box could be eliminated of have a timeout with a default of yes (keep caching). MP on the other hand just goes to all zoom levels without asking.

[image]

billbonney, 2015-05-07:
Great suggestion. The map caching dialog can be improved for sure.

lvale, 2015-05-08:
If improving the maps caching I would suggest something like:

Caching:
Zoom Start level: xx (number) yyyy (number of tiles - this can be retrieved rapidly from the maps provider)
Zoom Stop level : zz (number) wwww (number of tiles - this can be retrieved rapidly from the maps provider)

commit 1b22a3c5 (2015-07-08): Map Widget: Fixes #701 Add auto map caching option

Add option when caching maps to auto continue to a desired zoom level.
Form allows for both single level and multi-level caching and prompts
user at the end of either caching session to stop or continue.
closed 2015-07-08 by billbonney