ArduPilot/apm_planner #703: Unable to build master

AndKe, 2015-05-07:
Makefile:59587: recipe for target 'build-release/obj/MAVLinkDecoder.o' failed
make: *** [build-release/obj/MAVLinkDecoder.o] Error 1

all details here ;https://gist.github.com/AndKe/c30b44f0d4e32fc26efc

of course, the strangest thing is that nobody modified it lately..

billbonney, 2015-05-07:
Clean build?

AndKe, 2015-05-07:
Yes, I've even tried a fresh git clone before reporting.

billbonney, 2015-05-07:
I've just tested on Linux 32Bit and OSX 64 Bit with a clean build and it works as expected.

This looks suspiciously like a bad merge https://gist.github.com/AndKe/c30b44f0d4e32fc26efc#file-gistfile1-txt-L18 into your branch. Are you using your master?

you can fix master to be exactly a copy of upstream using these instructions, try option2

(if you want to 'backup master' on your branch you can do
git checkout master
git checkout -b master_backup 
and that will create a branch that is a 'copy' of your master to backup_master

https://gist.github.com/billbonney/76de0f1a2e79f3a8a1a5

AndKe, 2015-05-07:
I tried git checkout master  - it compiled, but not current version.
doing a git checkout upstream/master   fails to compile.   
(I do make clean every time while testing)  - what puzzles me , is that earlier today, I did:
git clone https://github.com/diydrones/apm_planner.git (to a new folder)
qmake qgroundcontrol.pro 
make -j8 
.. and failed too.

billbonney, 2015-05-07:
I have fix (needed to use my linux 64bit machine to see the issue)

commit d9ad5a73 (2015-05-07): MAVLink: Fixes #703 Ambiguous type cast for 64bit Systems
commit d79694f5 (2015-05-07): MAVLink: Fixes #703 Ambiguous type cast for 64bit Systems
commit c454c66a (2015-05-07): MAVLink: Fixes #703 Ambiguous type cast for 64bit Systems
closed 2015-05-07 by billbonney