ArduPilot/apm_planner #705: UDP over Wifi connection not working

zlite, 2015-05-12:
When connecting to the new 3DR Wifi radios (IP address 192.168.4.1, port 14550), UDP connections do not work (the radios work fine with Mission Planner and Tower).  Furthermore the connection dialog seems redundant, asking for the UPD port twice (both at the top screen and again on the IP address box) and showing that the connection is established when it isn't.

See the following progression:

Initial screen:
[image]

Changing to an IP address:
[image]

Correcting the port to 14550:

[image]

As a point of comparison, here's how it works in Mission Planner (where there are no issues):

[image]

billbonney, 2015-05-12:
One is the listening UDP on the host 14550, the other ip:port is a forced address on where to send responses. Normally, the listening UDP port receives a packet from a client, and then returns responses to ip:port in the received packet. There is not a connection, it's more start/stop listening. Lost packets are just lost.

I can tidy the dialog up to be more representitive.

I don't have any 3DR WiFi Radios to test with unfortunately.

zlite, 2015-05-12:
Ah. So what should the remote port be?  "192.168.4.1:????",

billbonney, 2015-05-12:
It looks like from the MP dialogs it is also 14550.

I think the issue is that in this scenario the remote end is also just waiting for something to be sent, and nothing has. We have AP2 listening on it's ip@14550 and your vehicle listening on it's ip on 14550, but neither is sending a message until it hears the other. Stalemate.

This almost requires a new connection type for AP2 where you declare the remote host and it' sends a HB message on 'connect'. I can look into that as well

zlite, 2015-05-12:
Yes, I think that's right. Because the remote Wifi radio is serving as an
Access Point, not a client, it needs to know what the IP address is of the
client (in this case the laptop APM Planner is running on), so it waits to
hear from that client before sending.

In the case of Tower, we originally had to run a little app called TermUDP
and send it some text to start the flow of data, and only then Tower would
work. Now Tower has been updated to send that automatically.

I'm guessing Mission Planner does the same thing (send a little data on the
UDP to start the flow back)

billbonney, 2015-05-12:
That little piece of data should be a Heartbeat message as per the MAVLink spec. I'll need to create a new UDP Connection Link that works with the logic you have described. Instead of listening it sends a Hearbeat message to a UDP port at an IP address on 'connect'.  ie port 14550, ip 192.168.4.1

billbonney, 2015-06-22:
Closed. Added new UDP Client connection

closed 2015-06-22 by billbonney