ArduPilot/apm_planner #712: Unexpected flow control options (serial port error in VMWare)

AndKe, 2015-05-20:
Please see screenshot - serial port fails in VMWare + windows guest. 
APM Planner 2.0.17
[image]

Same VM confirms normally working serial ports by using mission planner, and MiniOSD config too.

billbonney, 2015-05-20:
I'm not sure what I am looking at here, can you be more specific? Are you sure the com port has been released by the host OS, sometimes you need to stop the host OS grabbing the port before the VM, you can do this by setting the VM to grab it as soon as it appears.

AndKe, 2015-05-20:
Hi, when it comes to VMware Workstation, I am very sure about the status.
Let me say I have full control of where the USB device is connected, and it's connected - as a USB device, not a mapped serial port, to the guest OS.  (I am VCP-DCV certified)

The same serial device (FTDI adapter in this case)  works with a 3DR Radio on Mission Planner, or can be used to connect MinimOSD to it's config tool.   - But attempting to open the com port using APM Planner - fails. 

Please see attached screenshot (sorry, but that poor excuse for an OS does not allow copy from the terminal window whithout hopping thru loops, so taking a screenshot on the host was quicker.)

[image]

The interesting thing is: The same error appears for the serial link (FTDI+radio) in both radio-config mode, (where port is destroyed, and it fails)   - and the same error appears during usual connection - but connection is a success !

billbonney, 2015-05-20:
I can test it in windows vm later (using vbox) on OSX