ArduPilot/apm_planner #713: accel calibration is broken for plane (and rover?)

tridge, 2015-05-22:
accel calibration seems to be broken for plane. Plane has also removed the 1D accel cal option for the 3.3.0 release, which means APMPlanner2 users have no way to calibrate.
When I try to calibrate it just shows the 30s timeout and doesn't display any of the messages to go to the next step.
It does work fine for copter. Note that plane, copter and rover all now use the same accel calibration methods, so hopefully the fix is just to remove all the special cases.
this was with version 2.0.17

billbonney, 2015-05-25:
Thanks, I'll add a fix.

tridge, 2015-05-25:
thanks Bill!

closed 2015-05-26 by billbonney

ranger175, 2015-05-30:
Hi guys,

Is it possible to find out when this fix might be released?  Thank you very much.

Adam.

billbonney, 2015-05-30:
It's fixed. Download the latest daily build from http://firmware.diydrones.com/

ranger175, 2015-05-30:
Thanks Bill!

I didn’t know about that page.

I’m a little confused, was it an APM Planner or Arduplane fix?

Thanks,

Adam.

billbonney, 2015-05-30:
Both. One was to remove the 1D cal messages from ArduPlane and the other was to not enable sending it from AP2

billbonney, 2015-05-30:
Ps: all platforms support 3D cal now only. You can do that outside the frame if needed. There is a new AHRS trim if required

ranger175, 2015-05-30:
Thank you.

So do I need to update the APM Planner and the Arduplane firmware?

Thanks Bill!

billbonney, 2015-05-30:
Best to ask those questions on the forum http://ardupilot.com/forum/viewforum.php?f=1

commit 71ab1ab7 (2015-05-26): Accel Calibration: Fixes #713 Update Accel Calibration for 3D only for all vehicles