ArduPilot/apm_planner #715: libopenscenegraph80 dependency for Ubuntu-14-04

binarynut, 2015-05-24:
Where do I find dependency libopenscenegraph80 for install of Missionplanner on Ubuntu-14-04 ?

And I would imagine that libopenscenegraph80 would also have it's own dependencies.

binarynut, 2015-05-24:
I stumbled across libopenscenegraph80 on the internet and got it installed.

And now a new problem, when I run ./Qgroundcontrol I get the error below;

./qgroundcontrol: error while loading shared libraries: libQt5QuickWidgets.so.5: cannot open shared object file: No such file or directory

O will I ever get Qgroundcontrol working, I got a new build, Hexicopter and now I can't talk to it.

billbonney, 2015-05-25:
I'm confused. I am building APM Planner 2.0 fine on 32bit Linux Mint 17.1 (based on 14.04) with Qt5.2.1 installed.

do you have libqt5declarative5 installed?

harayz, 2015-11-09:
any update on this issue?

billbonney, 2015-11-09:
This error usually means you have the wrong version of Qt installed, or haven't installed the dependencies. See https://github.com/diydrones/apm_planner/blob/master/README.md

closed 2015-11-09 by billbonney