ArduPilot/apm_planner #716: High CPU usage on Linux

lostcontrol, 2015-05-28:
Hi,

I notice that Planner uses a lot of CPU cycles at some point when using the graph view. The high CPU load is not present when starting the application but only start after some time, I think after using the graph view a bit. Once the high CPU load starts, it never ends and I need to quit the application and restart it to get rid of it. This is drain my laptop battery pretty fast.

I tried to use Linux Perf to get some more information and it seems it is related to some GUI refresh operations. When Planner is eating CPU cycles like crazy, I also see that my X server has an high load. Unfortunately, I don't know enough of Qt to debug it more.

Am I the only one to see this?

I'm using latest master (but I always had this behavior as far as I can tell) compiled on Ubuntu 15.04.

Cyril

AndKe, 2015-06-04:
Not sure what kind of use of graph view you mean, I loaded a log , and it's fine.
check if maybe it's related to this one: https://github.com/diydrones/apm_planner/issues/646

lostcontrol, 2015-06-04:
Yes, I have to connect first (and then go to graph view) to make this bug appear. But even if I disconnect, the high CPU load remains. Without connecting first, I don't see this problem at all. What I do is that I connect, download the logs, quit Planner and restart it to analyze my logs.

I will provide an exact test case for this. Also would Linux Perf data be useful?

lostcontrol, 2015-06-05:
Sorry, I tried again and actually I noticed that right after connect, the CPU load start to increase as soon as I connect and didn't go away when disconnecting. Here is a extract of top (displaying threads also) after being disconnected:

```
11084 cyril     20   0 4861952 258072  86152 R 41.5  3.2   1:15.28 apmplanner2                                                                              
 1766 root      20   0  479500 108568  81236 S 31.9  1.4  11:33.17 Xorg                                                                                     
11085 cyril     20   0 4861952 258072  86152 R 18.6  3.2   0:28.39 QXcbEventReader         
```

I guess some events are bombarding the X server.

So actually I don't need to open the graph view to trigger this bug and after re-reading #646, I guess I face the same bug. Therefore I will close this ticket. Sorry.

closed 2015-06-05 by lostcontrol