ArduPilot/apm_planner #720: Copter PIDs: Improve Tuning Workflow

billbonney, 2015-05-31:
base on the comment here https://github.com/diydrones/apm_planner/issues/685#issuecomment-105181492

Add improved workflow for Extended Tuning for copter

lthall, 2015-05-31:
Hi Bill,
The filter frequencies are still missing from the Rate Loops.
I am thinking it is probably worth separating the loiter and Alt_Hold controller pid settings into another screen and making this one more focused on the tuning most people do. Gui design is hard so but I can help you with workflow.

Missing Rate loop settings
RATE_PIT_FILT_HZ
RATE_RLL_FILT_HZ
RATE_YAW_FILT_HZ

IMU Rate Filter
INS_GYRO_FILTER

Attitude input shaping. This defines what the copter feels like to fly.
ATC_ACCEL_P_MAX
ATC_ACCEL_R_MAX
ATC_ACCEL_Y_MAX
ATC_RATE_FF_ENAB
RC_FEEL_RP

These are the settings that define Acro.
ACRO_BAL_PITCH
ACRO_BAL_ROLL
ACRO_EXPO
ACRO_RP_P

Autotune settings
AUTOTUNE_AGGR
AUTOTUNE_AXES

Thrust linearisation. This is one of the first things that is set and maybe should be in initial setup.
MOT_CURR_MAX
MOT_SPIN_ARMED
MOT_THR_MIX_MIN
MOT_THST_BAT_MAX
MOT_THST_BAT_MIN
MOT_THST_EXPO
MOT_THST_MAX

billbonney, 2015-05-31:
The rate loops are there. This is the screen shot of what I have pushed so far. (download from here http://firmware.diydrones.com/Tools/APMPlanner/daily/2015-05-31/ )
[image]

billbonney, 2015-06-01:
PS: If you draw the layout (pencil and camera shot work) of the layout that works I can implement it.