ArduPilot/apm_planner #727: First spline leg incorrect

sgofferj, 2015-06-04:
The first spline leg is at least displayed incorrectly. I didn't try flying it yet.

sgofferj, 2015-06-04:
[image]

billbonney, 2015-06-04:
I think the issue is that your HOME point and WP1 have to be in the same location. I'll take a look at a fix, but that's the workaround. HOME needs to be moveable again, as AC now supports a moveable home point

AndKe, 2015-07-04:
Yes. Home should be movable. Also, it's a pain to have home at coordinates 0,0  - if 0,0  - then it should be ignored, or even made up near first item.
Whenever home spawns at 0,0  - for clean planning - it needs to be manually moved to realistic coordinates.